bigtreetech/BIGTREETECH-SKR-mini-E3 #320: Im trying to compile Marlin 2 for BlTouch with my offsets but cannot get it to work!

josh40996, 2020-07-03:
I've been trying for 2 days to recompile other folks firmware and also tried editing the stock marlin 2 software to no avail. I am using VSC and Platformio and I have no experience with this kind of software. I keep getting errors, mainly always linked to "libmaple/gpio.h" and "STM32F103RC_btt_512K" when trying to compile.

I was hoping somebody would be able to help me solve the issue, or compile the files for me? Much appreciated!

Here is my latest attempt 

> > Executing task in folder Marlin: C:\Users\Joshu\.platformio\penv\Scripts\platformio.exe run <

Processing STM32F103RC_btt_512K (platform: ststm32; board: genericSTM32F103RC; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 7.0.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 4.10900.200602 (1.9.0)
 - framework-cmsis 2.50501.200527 (5.5.1)
 - tool-stm32duino 1.0.2
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 21 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <LiquidCrystal> 1.5.0
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.7.1
|   |-- <SoftwareSerialM> 1.0.0
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|-- <Adafruit MAX31865 library> 1.1.0
|-- <LiquidTWI2> 1.2.7
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.8.0
|-- <SailfishLCD>
|-- <SailfishRGB_LED>
|   |-- <Wire> 1.0
|-- <SlowSoftI2CMaster>
|-- <SoftwareSerialM> 1.0.0
|-- <USBComposite for STM32F1> 0.91
|-- <EEPROM> 2.0.1
|-- <Wire> 1.0
Building in release mode
Compiling .pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\onboard_sd.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\src\src\HAL\shared\HAL_spi_L6470.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\src\src\HAL\shared\backtrace\backtrace.cpp.o
In file included from c:\marlin\marlin\src\hal\stm32f1\hal.h:36,
                 from c:\marlin\marlin\src\hal\hal.h:26,
                 from c:\marlin\marlin\src\inc\marlinconfig.h:30,
                 from Marlin\src\HAL\STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp:29:
c:\marlin\marlin\src\hal\stm32f1\fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp.o] Error 1
In file included from c:\marlin\marlin\src\hal\stm32f1\hal.h:36,
                 from c:\marlin\marlin\src\hal\hal.h:26,
                 from c:\marlin\marlin\src\inc\marlinconfig.h:30,
                 from Marlin\src\HAL\STM32F1\onboard_sd.cpp:15:
c:\marlin\marlin\src\hal\stm32f1\fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\onboard_sd.cpp.o] Error 1
In file included from c:\marlin\marlin\src\hal\stm32f1\hal.h:36,
                 from c:\marlin\marlin\src\hal\hal.h:26,
                 from c:\marlin\marlin\src\inc\marlinconfig.h:30,
                 from Marlin\src\HAL\shared\HAL_spi_L6470.cpp:28:
c:\marlin\marlin\src\hal\stm32f1\fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\shared\HAL_spi_L6470.cpp.o] Error 1
In file included from c:\marlin\marlin\src\hal\stm32f1\hal.h:36,
                 from c:\marlin\marlin\src\hal\hal.h:26,
                 from c:\marlin\marlin\src\inc\marlinconfig.h:30,
                 from c:\marlin\marlin\src\core\serial.h:24,
                 from Marlin\src\HAL\shared\backtrace\backtrace.cpp:29:
c:\marlin\marlin\src\hal\stm32f1\fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\shared\backtrace\backtrace.cpp.o] Error 1
================================================================================================== [FAILED] Took 5.89 seconds ==================================================================================================

Environment               Status    Duration
------------------------  --------  ------------
mega2560                  IGNORED
mega1280                  IGNORED
rambo                     IGNORED
FYSETC_F6_13              IGNORED
FYSETC_F6_14              IGNORED
sanguino644p              IGNORED
sanguino1284p             IGNORED
melzi                     IGNORED
melzi_optiboot            IGNORED
at90usb1286_cdc           IGNORED
at90usb1286_dfu           IGNORED
DUE                       IGNORED
DUE_USB                   IGNORED
DUE_debug                 IGNORED
LPC1768                   IGNORED
LPC1769                   IGNORED
STM32F103RC               IGNORED
STM32F103RC_fysetc        IGNORED
STM32F103RC_btt           IGNORED
STM32F103RC_btt_USB       IGNORED
STM32F103RC_btt_512K      FAILED    00:00:05.890
STM32F103RC_btt_512K_USB  IGNORED
STM32F103RE               IGNORED
STM32F103RE_btt           IGNORED
STM32F103RE_btt_USB       IGNORED
STM32F4                   IGNORED
STM32F7                   IGNORED
ARMED                     IGNORED
STM32F103VE_GTM32         IGNORED
STM32F103VE_longer        IGNORED
mks_robin_mini            IGNORED
mks_robin_nano            IGNORED
mks_robin                 IGNORED
mks_robin_pro             IGNORED
mks_robin_lite            IGNORED
mks_robin_lite3           IGNORED
jgaurora_a5s_a1           IGNORED
STM32F103CB_malyan        IGNORED
chitu_f103                IGNORED
STM32F401VE_STEVAL        IGNORED
FLYF407ZG                 IGNORED
FYSETC_S6                 IGNORED
STM32F407VE_black         IGNORED
BIGTREE_SKR_PRO           IGNORED
BIGTREE_GTR_V1_0          IGNORED
BIGTREE_BTT002            IGNORED
teensy31                  IGNORED
teensy35                  IGNORED
esp32                     IGNORED
linux_native              IGNORED
SAMD51_grandcentral_m4    IGNORED
rumba32_f446ve            IGNORED
rumba32_mks               IGNORED
include_tree              IGNORED
============================================================================================= 1 failed, 0 succeeded in 00:00:05.890 =============================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

eminentspoon, 2020-07-04:
I had similar issues to this when I was looking at getting the latest version of Marlin onto my SKR Mini E3 2.0 today. The issue is caused by PlatformIO changing their default core to STM32DUINO whereas Marlin needs maple. See [here](https://reprap.org/forum/read.php?415,874307) (issue 1) for the suggest fix which requires editing the platformio.ini file for the section in question (you'll need to update [env:STM32F103RC_btt_512K])). This issue has been fixed in the latest marlin bugfix-2.0x branch if you wanted to use that instead, copying over the specific config for your printer. Depending on the revision of the SKR mini E3 that you are using, you might have to use the bugfix branch anyway as the main branch doesn't have the details for the SKR mini E3 v2.0 revision.

josh40996, 2020-07-04:
Thank you!

I used the bug fix version and it worked flawlessly!