bigtreetech/BIGTREETECH-SKR-mini-E3 #327: Cannot build firmware Marlin-2.0.x-SKR-mini-E3-V2.0

mclemmon, 2020-07-13:
cannot open source file "stdint.h"

sam01508, 2020-07-13:
Hi mclemmon, The boards.h section needs updating also as the board does not yet appear did you find this also? It also fails on "USE_CONTROLLER_FAN requires a CONTROLLER_FAN_PIN. Define in Configuration_adv.h.
" #error "HEATER_0_PIN not defined for this board." 
 #error "E0_STEP_PIN, E0_DIR_PIN, or E0_ENABLE_PIN not defined for this board." 
  #error "TMC2208 or TMC2209 on X requires X_HARDWARE_SERIAL or X_SERIAL_(RX|TX)_PIN."
 Even Bigtreetech prebuilt firmware fails for me anyway how did this go for you.

abacabie, 2020-07-14:
I would use of the bugfix  version of Marlin that compiles for the V2, that should not give you any trouble.

sam01508, 2020-07-14:
abacabie, thanks for the tip, it does compile with a couple of yellow warning but it is successful. Are the messages written in yellow much to worry about?
Strange though I've never had a bugfix version compile successfully before.

abacabie, 2020-07-14:
Glad it worked for you, i never really look at the yellow warning lines, still very much a beginner.

mclemmon, 2020-07-14:
> 
> 
> I would use of the bugfix version of Marlin that compiles for the V2, that should not give you any trouble.

This is what I ended up doing. When I tried compiling a stock Marlin-2.0.x firmware, I got some of the same errors, as soon as I tried Marlin-bugfix-2.0.x I was able to get it to compile.