bigtreetech/BIGTREETECH-SKR-mini-E3 #328: [BUG] SKR E3 mini V2.0 CNC Fans always on

bazfp, 2020-07-13:
# Description
I have just finished wiring my SKR E3 mini V2.0 and my part-cooling fan and motherboard-case are constantly on. I have reconnected the fans in both on the motherboard and still the fans are always on. Once they start spinning I cannot control them via the menu or via M106. They just keep spinning.

**Expected behavior**
Be able to CNC control Fan0 and Fan1

**Actual behavior**
Fan0 is always on. When a stepper moves, Fan1 turns on and never turns off again.

Is this a board fault? MOSFET failure?

Andre6553, 2020-07-26:
i just install my board today and also noted this issue the part cooling fan is always on.

what i did i swap the mainboard fan and part cooling fan wires on the board.
to test and it work i can now control the fan speed from my lcd

lDeXTeDl, 2020-07-31:
Hi.
have same problem , nothing helps.
I think my SKR mini E3 v2 come broken!
2 x ports have 24 volt , as soon i start the board all coolers turn on , can't be controlled by soft / screen .......

Nincomp00p, 2020-08-05:
Elu.
Same problem here...
Tried rewiring, reflashed firmware, different fans, nothing helps... 
The Fan0 LED lights up when nothing is connected, however, when I connect a fan, it turns off and the fan doesnt work.
Fan1 goes 100% all the time, cannot regulate it...

lDeXTeDl, 2020-08-05:
> Elu.
> Same problem here...
> Tried rewiring, reflashed firmware, different fans, nothing helps...
> The Fan0 LED lights up when nothing is connected, however, when I connect a fan, it turns off and the fan doesnt work.
> Fan1 goes 100% all the time, cannot regulate it...

Now i know broblem ,, its is mosfet ! 
Burned or defective.
Do you see resistance on the mosfet ?

Mine is 

1 fan 1CNC pin1- 8ohm 
2 fan 1/CNC pin2- 11.5ohm

3 fan0 pin1- 80ohm
3 fan0 pin2- 3.5ohm

Nincomp00p, 2020-08-05:
Havent had time to check the mosfet yet, but on the to-do list.

lDeXTeDl, 2020-08-05:
> Havent had time to check the mosfet yet, but on the to-do list.

If you have the same +/- resistance, then is burned or defective. :/

skl111, 2020-08-26:
@lDeXTeDl What should the resistance be on each?

DaVaMa1, 2021-08-02:
I just installed my SKR mini V2 and as soon as I turned on the power a smell came off, opening up the case showed the mosfet blew immediatly. Had the exact same problem on the V1.2 (and on the Hotbed on that one). Going for a refund/replacement.