bigtreetech/BIGTREETECH-SKR-mini-E3 #329: [BUG] Thermal Runaway SKR Mini E3 v2.0

aztrip, 2020-07-14:
### Description
Connecting board to computer via usb causes low voltage error and thermal runaway until auto-shutdown.

### Steps to reproduce

1. [First Step] Connect Mac computer to SKR board via USB cable
2. [Second Step]Power on printer

**Expected behavior**

Access printer via computer

**Actual behavior**

Display beeps, shows low voltage error and finally thermal runaway occurs until auto-shutdown or power off.
### Additional Information
Sent same info to BTT, but no response.

abacabie, 2020-07-14:
I had something similar happen only I got a TMC error, it disappeared after I unplugged the cable and restarted the printer, good luck getting a reply from BTT, my mini V2 arrived faulty and they have ignored all my pleas for help.

aztrip, 2020-07-14:
BTT tech support has been great but I message them directly so that may be the difference