bigtreetech/BIGTREETECH-SKR-mini-E3 #330: Cannot build firmware Marlin-2.0.x-SKR-mini-E3-  (file not found?)

robstermn, 2020-07-14:
### FileNotFoundError:

When I try to compile I get the following error?

`> Executing task: C:\Users\r_vei\.platformio\penv\Scripts\platformio.exe run --environment STM32F103RC_btt_512K <

> Executing task: C:\Users\r_vei\.platformio\penv\Scripts\platformio.exe run <

Processing STM32F103RC_btt_512K (platform: ststm32; board: genericSTM32F103RC; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 6.0.0 > STM32F103RC (48k RAM. 256k Flash)    
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32-maple 1.10000.190819 (1.0.0)        
 - tool-stm32duino 1.0.2
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Looking for TMCStepper library in registry
Found: https://platformio.org/lib/show/5513/TMCStepper
LibraryManager: Installing id=5513 @ >=0.6.2,<1.0.0
Using cache: C:\Users\r_vei\.platformio\.cache\6b\892065df0a1fff8b39bce1802a2be26b
Unpacking...
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\r_vei\\Downloads\\BIGTREETECH-SKR-mini-E3-master (1)\\BIGTREETECH-SKR-mini-E3-master\\firmware\\V1.2\\Marlin-2.0.x-SKR-Mini-E3-V1.2\\.pio\\libdeps\\STM32F103RC_btt_512K\\_tmp_installing-_0u_pbfk-package\\.\\docs\\class_t_m_c2130_stepper_adde2041047c82600f5e01f52118f92a0_cgraph_org.svg':
  File "C:\Users\r_vei\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 168:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\r_vei\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\r_vei\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\r_vei\.platformio\platforms\ststm32\builder\main.py", line 98:
    target_elf = env.BuildProgram()
  File "C:\Users\r_vei\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\r_vei\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\r_vei\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\r_vei\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 131:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\r_vei\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\r_vei\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1036:
    project.install_dependencies()
  File "C:\Users\r_vei\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 868:
    lm.install(uri)
  File "C:\Users\r_vei\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 281:
    force=force,
  File "C:\Users\r_vei\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 704:
    pkg_dir = self._install_from_piorepo(name, requirements)
  File "C:\Users\r_vei\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 131:
    requirements,
  File "C:\Users\r_vei\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 482:
    self.unpack(dlpath, tmp_dir)
  File "C:\Users\r_vei\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 222:
    return fu.unpack(dest_dir, with_progress=False)
  File "C:\Users\r_vei\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 149:
    self._unpacker.extract_item(item, dest_dir)
  File "C:\Users\r_vei\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 82:
    super(TARArchive, self).extract_item(item, dest_dir)
  File "C:\Users\r_vei\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 39:
    self._afo.extract(item, dest_dir)
  File "C:\Users\r_vei\.platformio\python37\lib\tarfile.py", line 2044:
    numeric_owner=numeric_owner)
  File "C:\Users\r_vei\.platformio\python37\lib\tarfile.py", line 2114:
    self.makefile(tarinfo, targetpath)
  File "C:\Users\r_vei\.platformio\python37\lib\tarfile.py", line 2155:
    with bltn_open(targetpath, "wb") as target:
================================================================= [FAILED] Took 3.00 seconds =================================================================

Environment               Status    Duration
------------------------  --------  ------------
mega2560                  IGNORED
mega1280                  IGNORED
rambo                     IGNORED
FYSETC_F6_13              IGNORED
FYSETC_F6_14              IGNORED
sanguino644p              IGNORED
sanguino1284p             IGNORED
melzi                     IGNORED
melzi_optiboot            IGNORED
at90usb1286_cdc           IGNORED
at90usb1286_dfu           IGNORED
DUE                       IGNORED
DUE_USB                   IGNORED
DUE_debug                 IGNORED
LPC1768                   IGNORED
LPC1769                   IGNORED
STM32F103RC               IGNORED
STM32F103RC_fysetc        IGNORED
STM32F103RC_btt           IGNORED
STM32F103RC_btt_USB       IGNORED
**STM32F103RC_btt_512K      FAILED    00:00:03.002**
STM32F103RC_btt_512K_USB  IGNORED
STM32F103RE               IGNORED
STM32F103RE_btt           IGNORED
STM32F103RE_btt_USB       IGNORED
STM32F4                   IGNORED
STM32F7                   IGNORED
ARMED                     IGNORED
STM32F103VE_GTM32         IGNORED
STM32F103VE_longer        IGNORED
mks_robin_mini            IGNORED
mks_robin_nano            IGNORED
mks_robin                 IGNORED
mks_robin_pro             IGNORED
mks_robin_lite            IGNORED
mks_robin_lite3           IGNORED
jgaurora_a5s_a1           IGNORED
STM32F103CB_malyan        IGNORED
chitu_f103                IGNORED
STM32F401VE_STEVAL        IGNORED
FLYF407ZG                 IGNORED
FYSETC_S6                 IGNORED
STM32F407VE_black         IGNORED
BIGTREE_SKR_PRO           IGNORED
BIGTREE_GTR_V1_0          IGNORED
BIGTREE_BTT002            IGNORED
teensy31                  IGNORED
teensy35                  IGNORED
esp32                     IGNORED
linux_native              IGNORED
SAMD51_grandcentral_m4    IGNORED
rumba32_f446ve            IGNORED
rumba32_mks               IGNORED
include_tree              IGNORED
============================================================ 1 failed, 0 succeeded in 00:00:03.002 ==================

mclemmon, 2020-07-14:
I couldn't compile the firmware either using this repository. I had to use the Marlin-bugfix-2.0.x repo and build it using that.

brew99, 2020-07-14:
Try placing the downloaded files closer to the root directory of your harddrive.   It has a hard time with such long file paths

mclemmon, 2020-07-15:
> 
> 
> Try placing the downloaded files closer to the root directory of your harddrive (or remove some of the directories). It has a hard time with such long file paths

How much closer can you get than "C:\Users\r_vei.platformio"?

brew99, 2020-07-15:
> > Try placing the downloaded files closer to the root directory of your harddrive (or remove some of the directories). It has a hard time with such long file paths
> 
> How much closer can you get than "C:\Users\r_vei.platformio"?

What I'm saying is shorten the folder structure of the downloaded content.  You really don't need all this "BIGTREETECH-SKR-mini-E3-master (1)\BIGTREETECH-SKR-mini-E3-master\firmware\V1.2"

GhostlyCrowd, 2020-07-15:
Shorten your path like suggested above and VS code will need to be run as administrator to access the %userprofile% pathing for PIO lib deps.

robstermn, 2020-07-16:
I'll test that right now and report back.

robstermn, 2020-07-16:
Same error?

`--------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 6.0.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32-maple 1.10000.190819 (1.0.0)
 - tool-stm32duino 1.0.2 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Looking for TMCStepper library in registry
Found: https://platformio.org/lib/show/5513/TMCStepper
LibraryManager: Installing id=5513 @ >=0.6.2,<1.0.0
Using cache: C:\Users\r_vei\.platformio\.cache\6b\892065df0a1fff8b39bce1802a2be26b
Unpacking...
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\r_vei\\Desktop\\bigtree\\BIGTREETECH-SKR-mini-E3-master\\BIGTREETECH-SKR-mini-E3-master\\firmware\\V2.0\\Marlin-2.0.x-SKR-mini-E3-V2.0\\.pio\\libdeps\\STM32F103RC_btt_512K\\_tmp_installing-8_ql4euf-package\\.\\docs\\class_t_m_c2130_stepper_adde2041047c82600f5e01f52118f92a0_cgraph_org.svg':
  File "C:\Users\r_vei\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 168:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\r_vei\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\r_vei\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\r_vei\.platformio\platforms\ststm32\builder\main.py", line 98:
    target_elf = env.BuildProgram()
  File "C:\Users\r_vei\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\r_vei\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\r_vei\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\r_vei\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 131:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\r_vei\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\r_vei\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1036:
    project.install_dependencies()
  File "C:\Users\r_vei\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 868:
    lm.install(uri)
  File "C:\Users\r_vei\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 281:
    force=force,
  File "C:\Users\r_vei\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 704:
    pkg_dir = self._install_from_piorepo(name, requirements)
  File "C:\Users\r_vei\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 131:
    requirements,
  File "C:\Users\r_vei\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 482:
    self.unpack(dlpath, tmp_dir)
  File "C:\Users\r_vei\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 222:
    return fu.unpack(dest_dir, with_progress=False)
  File "C:\Users\r_vei\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 149:
    self._unpacker.extract_item(item, dest_dir)
  File "C:\Users\r_vei\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 82:
    super(TARArchive, self).extract_item(item, dest_dir)
  File "C:\Users\r_vei\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 39:
    self._afo.extract(item, dest_dir)
  File "C:\Users\r_vei\.platformio\python37\lib\tarfile.py", line 2044:
    numeric_owner=numeric_owner)
  File "C:\Users\r_vei\.platformio\python37\lib\tarfile.py", line 2114:
    self.makefile(tarinfo, targetpath)
  File "C:\Users\r_vei\.platformio\python37\lib\tarfile.py", line 2155:
    with bltn_open(targetpath, "wb") as target:
================================================================= [FAILED] Took 2.41 seconds ================================================================= 

Environment               Status    Duration
------------------------  --------  ------------
mega2560                  IGNORED
[... 29 lines cut ...]
mks_robin_mini            IGNORED
mks_robin_nano            IGNORED
mks_robin                 IGNORED
mks_robin_pro             IGNORED
mks_robin_lite            IGNORED
mks_robin_lite3           IGNORED
jgaurora_a5s_a1           IGNORED
STM32F103CB_malyan        IGNORED
chitu_f103                IGNORED
STM32F401VE_STEVAL        IGNORED
FLYF407ZG                 IGNORED
FYSETC_S6                 IGNORED
STM32F407VE_black         IGNORED
BIGTREE_SKR_PRO           IGNORED
BIGTREE_GTR_V1_0          IGNORED
BIGTREE_BTT002            IGNORED
teensy31                  IGNORED
teensy35                  IGNORED
esp32                     IGNORED
linux_native              IGNORED
SAMD51_grandcentral_m4    IGNORED
rumba32_f446ve            IGNORED
rumba32_mks               IGNORED
include_tree              IGNORED
===========`

robstermn, 2020-07-16:
I don't think its a folder issue as I'm able to compile an older version in that same directory?
See screen capture below?

[https://drive.google.com/file/d/1UJozaZozKFTdi-_6vw2hQaJmuj6KO_Gj/view?usp=sharing](url)

GhostlyCrowd, 2020-07-16:
FileNotFoundError: [Errno 2] No such file or directory: 'C:\Users\r_vei\Desktop\bigtree\BIGTREETECH-SKR-mini-E3-master\BIGTREETECH-SKR-mini-E3-master\firmware\V2.0\Marlin-2.0.x-SKR-mini-E3-V2.0\.pio\libdeps\STM32F103RC_btt_512K\_tmp_installing-8_ql4euf-package\.\docs\class_t_m_c2130_stepper_adde2041047c82600f5e01f52118f92a0_cgraph_org.svg

That path is WAY too long for windows. you need to move the source to a shorter path, put all the course in a folde on the root of c:\

Put the "Marlin-2.0.x-SKR-mini-E3-V2.0" folder on the c drive and then try again.

robstermn, 2020-07-16:
The previous post was in that same directory? And it worked? So it's not the path?

GhostlyCrowd, 2020-07-16:
> The previous post was in that same directory? And it worked? So it's not the path?

I cant see the previous post, the link is private.  

But regardless it probably worked because it wasn't exceeding the character limit of with paths.  Because I'm sure the older firmware used different libdeps

`C:\Users\r_vei\Desktop\bigtree\BIGTREETECH-SKR-mini-E3-master\BIGTREETECH-SKR-mini-E3-master\firmware\V2.0\Marlin-2.0.x-SKR-mini-E3-V2.0\.pio\libdeps\STM32F103RC_btt_512K\_tmp_installing-8_ql4euf-package\.\docs\class_t_m_c2130_stepper_adde2041047c82600f5e01f52118f92a0_cgraph_org.svg`

 is 283 character and the HARD limit for path characters is 260 in windows.

robstermn, 2020-07-16:
Okay different error?

`Processing STM32F103RC_btt_512K (platform: ststm32; board: genericSTM32F103RC; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 6.0.0 > STM32F103RC (48k RAM. 256k Flash)    
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32-maple 1.10000.190819 (1.0.0)        
 - tool-stm32duino 1.0.2
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 40 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <LiquidCrystal> 1.0.7 #780e0a1
|-- <TMCStepper> 0.4.5 #58c47ee
|   |-- <SoftwareSerialM> 1.0.0
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|-- <Adafruit MAX31865 library> 1.2.1
|   |-- <Adafruit BusIO> 1.3.3
|   |   |-- <Wire> 1.0
|-- <LiquidTWI2> 1.2.7
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.8.0
|-- <SlowSoftI2CMaster>
|-- <SoftwareSerialM> 1.0.0
|-- <USBComposite for STM32F1> 0.91
|-- <STM32ADC> 1.0
|-- <EEPROM>
|-- <Wire> 1.0
Building in release mode
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduinoVariant\board.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduinoVariant\wirish\boards.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduinoVariant\wirish\boards_setup.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduinoVariant\wirish\start.S.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduinoVariant\wirish\start_c.c.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduinoVariant\wirish\syscalls.c.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\HardwareTimer.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\Print.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\WString.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\avr\dtostrf.c.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\cxxabi-compat.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\ext_interrupts.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\hooks.c.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\itoa.c.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\libmaple\adc.c.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\libmaple\adc_f1.c.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\libmaple\bkp_f1.c.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\libmaple\dac.c.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\libmaple\dma.c.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\libmaple\dma_f1.c.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\libmaple\exc.S.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\libmaple\exti.c.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\libmaple\exti_f1.c.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\libmaple\flash.c.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\libmaple\fsmc_f1.c.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\libmaple\gpio.c.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\libmaple\gpio_f1.c.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\libmaple\i2c.c.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\libmaple\i2c_f1.c.o
[... 679 lines cut ...]
mks_robin_lite            IGNORED
mks_robin_lite3           IGNORED
jgaurora_a5s_a1           IGNORED
STM32F103CB_malyan        IGNORED
chitu_f103                IGNORED
STM32F401VE_STEVAL        IGNORED
FLYF407ZG                 IGNORED
FYSETC_S6                 IGNORED
STM32F407VE_black         IGNORED
BIGTREE_SKR_PRO           IGNORED
BIGTREE_GTR_V1_0          IGNORED
BIGTREE_BTT002            IGNORED
teensy31                  IGNORED
teensy35                  IGNORED
esp32                     IGNORED
linux_native              IGNORED
SAMD51_grandcentral_m4    IGNORED
rumba32_f446ve            IGNORED
rumba32_mks               IGNORED
include_tree              IGNORED
============================================================ 1 failed, 0 succeeded in 00:03:00.085 ============================================================
The terminal process "C:\Users\r_vei\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
`

GhostlyCrowd, 2020-07-16:
That looks liek its compiling now and is failing on adafruit I2C. if youre using an i2c eeprom you might want to consider compiling the latest marlin bugfix source as this will probably be resolved in it.

brew99, 2020-07-16:
Or see this post:
[https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/258](url)

abacabie, 2020-07-17:
As has been said just use the bugfix, it compiles fine and seems issue free.