bigtreetech/BIGTREETECH-SKR-mini-E3 #332: Bricked board?

magilla5592, 2020-07-16:
Hi All, Just got the skr mini e3 V2.0 and installed in my Ender 3.  Worked fine.  I downloaded the latest .bin from the here and put it on the SD and rebooted.  Now, I get a blue screen and nothing else.  I am unable to connect by USB, either.  

It does appear to be flashing, as the board blinks and the .bin turns into a .cur, but I cannot otherwise engage with the board.  I'm a noob, so I'm still trying to figure out the platformio and compiling stuff. 

Does anyone have a .bin or a suggestion on how I can start using this board again?  

Thanks!

nathanandtech, 2020-07-20:
Same problem here, did you find a fix? Thanks

magilla5592, 2020-07-20:
No, not yet.  I'm still exchanging emails with BTT support.

nathanandtech, 2020-07-20:
I just flashed this: https://github.com/gazcbm/Marlin-2.0.x-SKR-Mini-E3-v1.2/blob/master/CompiledFirmWare/firmware.bin
By putting it on an SD card, and switching the printer on. After a few seconds it came on. So I think the firmware here is not working. 
Unfortunately that build doesn't have manual bed leveling, so i'll have to build my own eventually. But at least the screen comes on now!

magilla5592, 2020-07-20:
Tried that file.  Closer!  I was finally able to connect via USB, but I plugged it into the printer, the display is still all blue...  

At this point, I just want o go back to how it came out of the box....

nathanandtech, 2020-07-20:
That's good, so just to understand, you turn the printer on, with the sd card and usb cable unplugged, and nothing shows on the screen, just the blue backlight?

magilla5592, 2020-07-20:
Not quite.  I started with just the board on my desk.  I put the .bin on the SD and powered the board via USB and it ran through and updated (.bin to .cur).  I then was able to connect to via USB (through pronterface).  I was hoping that it was successful, so I then took the board and put it in my printer- just 24v main and display connected.  When I powered the printer on, the display was just blue.

erichards-nvidia, 2020-09-01:
Did you by chance plug the lcd ribbon cable in to the wrong expansion port on the LCD when you moved from the desk to the printer?
My screen was blue with just a cursor when I had it wrong when I first assembled my Ender (stock board).