bigtreetech/BIGTREETECH-SKR-mini-E3 #334: Can't compile your BIGTREETECH-SKR-mini-E3/firmware/V2.0/Marlin-2.0.x-SKR-mini-E3-V2.0/Marlin/

zonalimitatore, 2020-07-20:
Try different time with platformio....nothing to do.I receive this:

Processing STM32F103RC_btt_512K (platform: ststm32; board: genericSTM32F103RC; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 7.0.0 > STM32F103RC (48k RAM. 256k Flash)    
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 4.10900.200602 (1.9.0)
 - framework-cmsis 2.50501.200527 (5.5.1)
 - tool-stm32duino 1.0.2
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing Adafruit_NeoPixel
Adafruit NeoPixel @ b72627a is already installed
LibraryManager: Installing U8glib-HAL
U8glib-HAL @ 0.4.1 is already installed
LibraryManager: Installing Adafruit_MAX31865
Adafruit_MAX31865 @ 1.1.0 is already installed
LibraryManager: Installing LiquidTWI2
LiquidTWI2 @ 1.2.7 is already installed
LibraryManager: Installing Arduino-L6470
Arduino-L6470 @ 0.8.0 is already installed
LibraryManager: Installing SailfishLCD
SailfishLCD @ 0.0.0 is already installed
LibraryManager: Installing SailfishRGB_LED
SailfishRGB_LED @ 0.0.0 is already installed
LibraryManager: Installing SlowSoftI2CMaster
SlowSoftI2CMaster @ 0.0.0 is already installed
LibraryManager: Installing SoftwareSerialM
SoftwareSerialM @ 1.0.0 is already installed
Looking for USBComposite_stm32f1 library in registry
Warning! Library `{'name': 'USBComposite_stm32f1', 'requirements': '==0.91'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'name': 'USBComposite_stm32f1', 'requirements': '==0.91'}` is a built-in library (included in framework, SDK). 
E.g., SPI, Wire, etc.
Found 20 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <LiquidCrystal> 1.5.0
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.7.1
|   |-- <SoftwareSerialM> 1.0.0
|-- <Adafruit NeoPixel> 1.2.4 #b72627a
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|-- <Adafruit MAX31865 library> 1.1.0
|-- <LiquidTWI2> 1.2.7
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.8.0
|-- <SailfishLCD>
|-- <SailfishRGB_LED>
|   |-- <Wire> 1.0
|-- <SlowSoftI2CMaster>
|-- <SoftwareSerialM> 1.0.0
|-- <EEPROM> 2.0.1
|-- <Wire> 1.0
Building in release mode
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduinoVariant\PeripheralPins.c.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduinoVariant\variant.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\HardwareTimer.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\Print.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\RingBuffer.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\FrameworkArduino\USBSerial.cpp.o
[... 332 lines cut ...]
mks_robin_mini            IGNORED
mks_robin_nano            IGNORED
mks_robin                 IGNORED
mks_robin_pro             IGNORED
mks_robin_lite            IGNORED
mks_robin_lite3           IGNORED
jgaurora_a5s_a1           IGNORED
STM32F103CB_malyan        IGNORED
chitu_f103                IGNORED
STM32F401VE_STEVAL        IGNORED
FLYF407ZG                 IGNORED
FYSETC_S6                 IGNORED
STM32F407VE_black         IGNORED
BIGTREE_SKR_PRO           IGNORED
BIGTREE_GTR_V1_0          IGNORED
BIGTREE_BTT002            IGNORED
teensy31                  IGNORED
teensy35                  IGNORED
esp32                     IGNORED
linux_native              IGNORED
SAMD51_grandcentral_m4    IGNORED
rumba32_f446ve            IGNORED
rumba32_mks               IGNORED
include_tree              IGNORED
===================================================== 1 failed, 0 succeeded in 00:00:39.885 =====================================================The terminal process "C:\Users\Bax\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

jtrmal, 2020-07-20:
bigtreetech/BIGTREETECH-SKR-mini-E3#302

zonalimitatore, 2020-07-20:
I've try to follow that,but return this:

In file included from Marlin\src\HAL\STM32F1\dogm\../../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\STM32F1\dogm\../../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\STM32F1\dogm\../../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp:29:
Marlin\src\HAL\STM32F1\dogm\../../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio\build\STM32F103RC_btt_512K\src\src\HAL\shared\esp_wifi.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\src\src\HAL\shared\servo.cpp.o
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\STM32F1\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\STM32F1\onboard_sd.cpp:15:
Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\onboard_sd.cpp.o] Error 1
In file included from Marlin\src\HAL\shared\../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\shared\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\shared\HAL_spi_L6470.cpp:28:
Marlin\src\HAL\shared\../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
In file included from c:\users\bax\desktop\marlin-bugfix-2.0.6\marlin\src\hal\stm32f1\hal.h:36,
                 from c:\users\bax\desktop\marlin-bugfix-2.0.6\marlin\src\hal\hal.h:26,
                 from Marlin\src\HAL\shared\backtrace\../../../core/../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\shared\backtrace\../../../core/serial.h:24,
                 from Marlin\src\HAL\shared\backtrace\backtrace.cpp:28:
c:\users\bax\desktop\marlin-bugfix-2.0.6\marlin\src\hal\stm32f1\fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\shared\backtrace\backtrace.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\shared\HAL_spi_L6470.cpp.o] Error 1
In file included from Marlin\src\HAL\shared\backtrace\../../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\shared\backtrace\../../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\shared\backtrace\../../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\shared\backtrace\unwmemaccess.cpp:18:
Marlin\src\HAL\shared\backtrace\../../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\shared\backtrace\unwmemaccess.cpp.o] Error 1
In file included from Marlin\src\HAL\shared\../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\shared\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\shared\eeprom_if_i2c.cpp:28:
Marlin\src\HAL\shared\../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\shared\eeprom_if_i2c.cpp.o] Error 1
In file included from Marlin\src\HAL\shared\../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\shared\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\shared\eeprom_if_spi.cpp:28:
Marlin\src\HAL\shared\../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\shared\eeprom_if_spi.cpp.o] Error 1
In file included from Marlin\src\HAL\shared\../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\shared\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\shared\esp_wifi.cpp:23:
Marlin\src\HAL\shared\../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\shared\esp_wifi.cpp.o] Error 1
In file included from Marlin\src\HAL\shared\../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\shared\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\shared\servo.cpp:54:
Marlin\src\HAL\shared\../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\shared\servo.cpp.o] Error 1

sam01508, 2020-07-21:
Hi, I've been through this for the version 2 board. Are you using the latest Bugfix version of marlin and the Bugfix config file? Thats what fixed it for me.
The No such file or directory error I had when the st stm32 updated from 6.1.1 to 7.1.1 but this also has now been fixed, but I copied and pasted the earlier platform in the platform ini file so it read : 
[env:STM32F103RC_btt_512K]
platform          = ststm32@<6.2.0
board             = genericSTM32F103RC

DRCUE2004, 2020-08-15:
It is same problem.
I have tried all the solutions that I have read ... it does not compile

https://ibb.co/brSCWhH

zonalimitatore, 2020-09-19:
Solved thanks

closed 2020-09-19 by zonalimitatore