bigtreetech/BIGTREETECH-SKR-mini-E3 #335: Screen in Marlin mode gets scrambled when starting a print

Whiskaya, 2020-07-20:
### Description

Screen scrambles text and shows unintelligible symbols that change fast when starting a print.
Sometimes this happens when starting in TFT mode and then switching to Marlin mode, without starting a print.
Sometimes text scrambles lightly in Marlin mode. Moving the knob may sometimes fix it.

### Steps to reproduce

1. Start printer normally (in Marlin mode).
2. Start preheat.
3. Start print.

**Expected behavior**
Normal printing.

**Actual behavior**
Printer keeps working as normal.
Entering menus (if you remember the moves on the knob) is still possible.
Screen is unreadable.

### Actions taken
1. Flashed stock (stable) firmware on main board and TFT board.
2. Changed ribbon cable.
3. Tried different SD cards, in all formats (FAT16 and FAT32) and sizes (micro w/ adapter and regular), from board and TFT mount.
4. Connected stock display.

### Hardware
Ender 5
Stock PSU
SKR Mini E3 V2.0
TFT35 E3 V3.0

First 2 pictures show "light scrambling" (recoverable when moving the knob).
Last 2 show "severe scrambling" (can't see the Marlin screen until reset).
[image]
[image]
[image]
[image]
[Videos.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4949805/Videos.zip)

jphansendk, 2020-07-26:
Have same issue.. Haven’t found solution yet.. But For some reason I have a suspicion to low memory on EEPROM is causing it.. just don’t know what to pick to shrink my Marlin firmware..

Whiskaya, 2020-07-26:
I've read some comments blaming the generic PSU, as soon as they changed it for a MeanWell, the problem was gone.
Maybe it's an interference problem, as it starts when heating... mostly.
Still, very few times it starts without any high current active (i.e. idling).
Tried many firmware changes so far, just to try and confirm if it's a hardware thing.

jphansendk, 2020-07-26:
Do you have a long cable like me connecting the screen and the board? Don’t know if that can be a part of the issue.. my “glitch” starts as an example if I run a PID calibration and go in to the menu afterwards to store the new PID.. that is impossible for me.. the glitch happines as soon I enter the menu.

Whiskaya, 2020-07-26:
I’m just using the cable that comes with the TFT35 V3.0.
For now you might want to try doing the PID from a computer and save it with M500. I've noticed even though the screen goes crazy, the printer itself doesn't stop working.

jphansendk, 2020-07-26:
Yeah guess so but it doesn't solve the issue 😄

jphansendk, 2020-07-26:
Just found this: https://marlinfw.org/docs/basics/troubleshooting.html

Whiskaya, 2020-07-26:
Nope... Still the same :/
Tried to fiddle around with the numbers. No luck.
At this point I'm 90% sure it's something to do with some kind of interference.

[IMG_9588.mov.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4978930/IMG_9588.mov.zip)
[IMG_9587.mov.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4978931/IMG_9587.mov.zip)

Whiskaya, 2020-07-26:
Just to make it more interesting... here's both displays at the same time(TFT35 using TFT cable and stock LCD using the serial one). No problems in the stock display with and without the firmware edit.
Seems as it could be a thing within the TFT35 board maybe?
[image]
[image]

jphansendk, 2020-07-27:
Or maybe the simulation causes the problem. As I understand it tft35 emulate the marlin mode? So maybe it's some settings in the tft35 firmware where the solution can be found?

jphansendk, 2020-07-27:
From a thread on facebook I made got this reply: Hi

I had exactly the same issue when I retrofitted one od my printers.  I nearly lost my mind trying to figure it out.  In my case, I clipped a ferrite bead around the TFT and EXP1 and 2 cables and the isue went away.  In my case the symptom only showed up when I turned on the bed heater.

jphansendk, 2020-07-27:
I'm not a code guy and don't have the guts to jump in right now but the new release of Marlin 2.0.6 could be a solution?

jphansendk, 2020-07-27:
Just compared my configuration.ini with a guy who had it working without the glitch.. the only difference is his baudrate is set to 250000 mine is set to 115200. Will test later

jphansendk, 2020-07-27:
Just tested the baudrate. Changed from 115200 to 250000 and I think that fixed the problem.. SOOO f****** simple.. will report back if I'm wrong.

Whiskaya, 2020-07-27:
Already tried Marlin 2.0.6 and baudrate to 250000... no luck.
Now I dare to say: ***It's most definitely a TFT board thing!***

vicewize, 2020-07-28:
I’ve been fighting this since day one. Tons of different Marlin flashes, TFT flashes, etc. I 100% believe it has to do with the TFTs hardware when trying to emulate. It happens to me if I try to edit like z-steps or anything and I’m turning encoder a lot it’ll happen, but then I’ve had it happen just within a menu. Can’t figure it out. So I use the stock display and the TFT for now.

jphansendk, 2020-07-29:
Do you have the baudrate set to 250000?

vicewize, 2020-07-29:
> Do you have the baudrate set to 250000?

I’ve tried mine with both Baud rates and same result. 115200 & 250000. It still occurs. I’m stumped.

Andyman14159, 2020-12-02:
Hi I'm commenting to keep this issue alive. December 2020 and I have tried all of these fixes with no good sustainable result. Has anyone came across anything new?

Beekeeper007, 2020-12-19:
same Problem with SKR Mini and tft35 v3 :(

Pipstapee, 2020-12-24:
I have the same issue after swapping over the hotend to a Volcano and extending the display serial cable to have the PSU at the back of my Ender 5. I've changed the hotend back to the standard one and it's still an issue so I'm going to assume it's the serial cable. I've a huge reconfig going on but will have a swap around to see what makes the difference and or solves the issue.

Pipstapee, 2020-12-28:
I swapped back to the original Ender screen and it worked fine with no issues, indicating it was the an issue with the TFT35. I remember having an issue with the BMPs when I flashed the firmware initially so revisited that. Although I didn't use the touchscreen icons I thought there may have been a connection some how. I flashed the TFT35 firmware again and followed this guide to the letter: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware
No issues after a few days of printing.

kermus, 2021-02-11:
I can say 90% certainty that the culprit is the motherboard (SKR Mini E3 v1.2) I have changed my own Ender3 power supply, display, all the display cable and firmware several times and each time the same problem comes on for about 30min-2 hours from start printing

Yathani, 2021-11-27:
Make sure the ribbon cables for the TFT are not touching any stepper drivers on the motherboard. Keep them away as much as you can.

TGTools79, 2023-02-13:
Hi there!
1st, thank you all very much for your comments as it helped me eliminate a bunch of possibilities. I (finally) managed to stop my TFT24 from doing the same. Just by supplying 5v to the "5v Independent Power Supply" port. It could be in combination with other stuff i did but as soon as I did this, the problem has vanished completely.
[image]

LouDnl, 2023-10-14:
After reading what @TGTools79 described here, I was able to resolve my issue with the random characters and the display freezing.

What I did was:
- Add a ferrite bead around the Pi4 power cable close to the USB-C input. 
  I previously had the Pi4 powered via printers power supply when the printer turned on and the issues appeared after I moved to an external power adapter.
- Reroute the 3 display cables around my Pi4 instead of over the Pi4.
- Add a ferrite bead around the display tft serial cable close to the mainboard
- Move the USB cable from a USB3 port to a USB2 port on the Pi4
  USB3 is known for causing interference at around 2.4Ghz (especially on GPS modules), see explanation here: https://www.usb.org/sites/default/files/327216.pdf
  this could have been causing some issue with the serial cable, dont know for sure but you never know ofcourse

Now it seems solved, will post here if not :)

closed 2024-12-21 by Whiskaya