bigtreetech/BIGTREETECH-SKR-mini-E3 #336: Z probe offset does nothing.

thedaviator, 2020-07-20:
Forgive me if this is the wrong place to post or if I sound like a moron but I'm at my wits end. 

I have the BIGTREETECH-SKR-mini-E3 board for my Ender 3. I flashed the firmware titled **firmware-bltouch-for-z-homing.bin** 
The BLTouch *seems" to do what it's supposed to do on startup. Clicks up and down and down and then ends up retracted and turns red. 
When I set my printer to auto home, it goes near the middle of the bed then goes up and down and comes as close as about 1-2 mm when the probe touches before it rises up to about 8-10 mm and stops. 

Then when I try and print, it's too high above the bed by about the same 1-2 mm. I tried adjusting the Probe Z offset but that does absolutely nothing. No matter how far I go negative or positive AND store the setting it does not move up or down. 

Also, how does the BLTouch work if it only extends when homing the printing? 

Thanks!

brew99, 2020-07-21:
Are you using the G29 (auto bed level) in your slicer?  This needs to be in your start g-code for it to do the probing of the bed after the homing (G28).

Also, you can try adjusting the baby stepping of the Z while it is printing to bring the nozzle closer to the bed

thedaviator, 2020-07-21:
Thanks but I found a video that showed how to calibrate BLTouch and then I found the code (G29 and more to prevent the oozing).

closed 2020-07-21 by thedaviator

grantlesueur, 2020-09-18:
Same issue for me.  ABL does the bed leveling and I would normal use z offset adjust as the first layer is going down.  As I adjust the z offset i would normally feel the z-shaft stepping as I make the adjustments..  Similarly I do not feel the shaft making adjustment steps so it unclear to me if the ABL is working.

I am using using the precompiled firmware.bin.   It would be nice if the source was avallablable to inspect these..

RedCoder09, 2020-09-22:
Yeah, I fixed the issue by decreasing the homing height and probing height, and before that, I lowered the endstop itself.