bigtreetech/BIGTREETECH-SKR-mini-E3 #337: [BUG] SKR mini E3 V2.0 Can't flash firmware

Jhyrachy, 2020-07-22:
### Description
The board does NOT accept any "firmware.bin" and just refuse to boot when one is inserted in the SD card reader

### Steps to reproduce

1. compile a "firmware.bin" or download one [from the official BTT page](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V2.0)
2. rename it to "firmware.bin" all lowercase
3. insert the micr-sd card in to the printer
4. turno on the printer

**Expected behavior**
Board should flash the "firmware.bin" and rename it to "firmware.CUR"

**Actual behavior**
The board does NOT boot, does NOT flash the firmware, does NOT rename it to firmware.CUR

### Additional Information
Configuration.h : https://pastebin.com/raw/Z4156QHf
Configuration_adv.h : https://pastebin.com/raw/1gg5uYgK

brew99, 2020-07-23:
try a different SD card (formatted as FAT32) and put the firmware.bin on it.

sentrid, 2020-08-31:
I also have this issue.  Boots up without updating firmware from firmware.bin file on the SD card.

Roguetrooper2k1, 2020-08-31:
Contact: Support@bigtree-tech.com 
They get you to jump through hoops, trying allsorts of things that you already have done. They now want a video of trying to flash firmware. they will then consider refunding you or resending the motherboard.
I live in hope...

Jhyrachy, 2020-08-31:
yes, I sended them a 10 minutes video of the flashing process to get a new one

sentrid, 2020-08-31:
I haven't had a chance to test this yet, but I am wondering if the scenario I describe below might be affecting things.  The problem that I'm experiencing is that the firmware is not updating when I put an SD card with "firmware.bin" file on it into the board.
I'm working with the BTT SKR mini E3 V2.0 and have plugged in a BLTouch where the 3 wire dupont connector is plugged into the left 3 Z-Probe pins, and the other 2 wires plugged into the Z-Stop pins.  Note that I had to alter the wiring in both the DuPont connector and the Z-Stop connector.  
Does anyone else have things wired this way and experiencing the same problem?

stangjr, 2020-09-22:
I am having the same issue. 
I've tried 4 different micro SD cards, formatted them all. Compile my firmware just fine and copy it to the SD card, re-name it to firmware.bin > plug it into the board > power on...... no flashing takes place. The board boots like normal and i can see the test gcode on the SD card.... so i am able to see the SD card.
I am now on my second board and having the same issue... Exactly the same issue. What am i doing wrong????

Roguetrooper2k1, 2020-09-23:
Instead of copying th bin file onto the sd card , move it. As I was doing the video to send BIQU  it flashed and that was the only thing  that I changed. This worked for me . Hope it does for you.

stangjr, 2020-09-24:
@Roguetrooper2k1 thanks for the tip but unfortunately it was un-successful. 
Im really at a loss here.  I've reached out to Bigtreetech support and they are useless.

Kal318, 2020-10-05:
> @Roguetrooper2k1 thanks for the tip but unfortunately it was un-successful.
> Im really at a loss here. I've reached out to Bigtreetech support and they are useless.

I am very interested in your results as I'm, having the same issue. I've had my board for about 3 weeks and it's starting to look like I'll have to replace it already. This problem is only compounded by the fact I also purchased a BL Touch from Biqu and I can't use it because I can't update the firmware. also my part cooling fan connection does not appear to be working either.

quadmaster55, 2020-10-07:
I am having the same issue, I am now on my third board. My first board actually did flash but the bl touch wasn't working properly so I tried different firmware and it bricked, my second board didn't flash at all and now I'm on my third board, the same issue, won't flash, just boots up as normal. I am fed up with this and want to start printing again, looking for any kind of help, please

Kal318, 2020-10-07:
> I am having the same issue, I am now on my third board. My first board actually did flash but the bl touch wasn't working properly so I tried different firmware and it bricked, my second board [...]

I've sent an email to there support line today. Just waiting on a word back.

quadmaster55, 2020-10-07:
> I've sent an email to there support line today. Just waiting on a word back.

Looking forward to hearing the results, please let me know so I can be stress free

tbcseod, 2020-10-07:
so i'm not the only apple under this tree....
I'm still what id call new to 3d printing and been doing some upgrades and this has just had me stumped for 3 days now. At first i figured i didn't understand marlin and compiling - then thought it was the new TFT35 screen...... but my brand new board is acting like it doesn't know it even has a sd slot......

quadmaster55, 2020-10-07:
> so i'm not the only apple under this tree....
> I'm still what id call new to 3d printing and been doing some upgrades and this has just had me stumped for 3 days now. At first i figured i didn't understand marlin and compiling - then thought it was the new TFT35 screen...... but ive sorted out both of those as not the issue nd it seems my brand new board is just acting like it doesn't know it even has a sd slot......

That's The same with me. I am able to print but I can't get any firmware to upload

Kal318, 2020-10-07:
Mmmm this is actually starting to sound like a quality control issue.

quadmaster55, 2020-10-07:
> Mmmm this is actually starting to sound like a quality control issue.

Is it fixable? Or is it a hardware problem

Kal318, 2020-10-07:
I personally don't like resoldering components to a PCB so I'm going to say no. This opinion is theory based on what's been said in this thread. I have not heard from Biqu yet.

quadmaster55, 2020-10-07:
> I personally don't like resoldering components to a PCB so I'm going to say no. This opinion is theory based on what's been said in this thread. I have not heard from Biqu yet.

Oh, okay, so since it's a hardware problem what do I do, just keep ordering and returning board from Amazon til I get a working one?

Kal318, 2020-10-09:
I'm not. I'm think about switch to the Creality 32 silent board actually. That's if Biqu doesn't get back to me soon.

quadmaster55, 2020-10-09:
> I'm not. I'm think about switch to the Creality 32 silent board actually. That's if Biqu doesn't get back to me soon.

Creality has a new board that they are putting in their ender 3 V2 which will also work in the ender 3 pro and that's what I would recommend getting, I'm thinking about getting one myself if I can't figure out this big tree tech problem

Kal318, 2020-10-09:
> Creality has a new board that they are putting in their ender 3 V2 which will also work in the ender 3 pro and that's what I would recommend getting, I'm thinking about getting one myself if I can't [...]

Does that version have active thermal Brun away protection? That's actually one of the main reasons I wanted the skr mini.

quadmaster55, 2020-10-09:
Doesn't say on their website, but it is the silent main board 4.2.7

Kal318, 2020-10-16:
Got the Creality board and everything work. 😊

quadmaster55, 2020-10-16:
> Got the Creality board and everything work. 😊

Yep, I just got my money back for the SKR board and my creality board should be here tomorrow

Kal318, 2020-10-16:
Did you order your skr directly from Biqu?

quadmaster55, 2020-10-16:
Not sure what biqu is but I ordered it from Amazon

aricut, 2020-10-30:
hi
had the same issue ,change the file name into FIRMWARE.BIN (capital letters) ,worked for me...

slyticoon, 2020-11-03:
> 
> 
> hi
> had the same issue ,change the file name into FIRMWARE.BIN (capital letters) ,worked for me...

I tried your solution, and it worked!

MatthewKing99, 2020-12-21:
Ive Figured it out finally, you have to enable extensions in windows 10. Just look up how to do that and you shouldnt have to rename the file to Firmware.bin, it should already be set to Firmware.bin

shawnseveras, 2020-12-26:
i went through that whole process now and it still didnt change the firmware version. and i used straight from marlin.

Boundless-Code, 2020-12-31:
I have the same problem on two different boards. None of the work arounds here work. I am done with this board. Makes me never want to buy another BTT motherboard again. Do the 1.4s have this problem also?

Marvincolon, 2021-01-01:
Hi all, I've been having this issue and have been working on it for more that 10 hours now. I have accidentally solved my problem and decided to post it here in hopes that it works for someone else. I tried everything mentioned here and nothing worked. I am using the E3 V2.0 board and the TFT35 touchscreen display, Got it from amazon as a bundle, Im thinking that some or most of everyone here probably got the same thing. I also added the BL touch so for me it was not working as well. My accidental fix for me was when I removed my display (printer was OFF and felt a little static shock so be careful) In the process of handling it I hit the reset button on the display, so I decided to just hold it down a few seconds while everything was off, then I restarted the printer. I was immediately greeted with BL touch errors which I have not seen before. Hit the OK that popped up on the screen and the printer did an auto level,  After this I am now seeing both SD slots, the one on the main board and the one on the display. Hope this helps someone.

shawnseveras, 2021-01-01:
so im hoping this finds everyone well. the solution that has worked for me is really simple but also a duh moment. so the problem isn't with he board,nor the firmware.   the problem is windows. when you rename the file to firmware.bin it is actually coming out to firmware.bin.bin that's because when you open the folder in windows if you click on the tab view,you will see 3 boxes unchecked hopefully. they will be item check boxes file name extensions and hidden items. you want to have file name extensions checked. if you already did the rename to firmware.bin then you will see what i mentioned. anyways this solved the problem for me completely. anyways i will post this to all the forums so that way everyone knows. mainly because i found the solution by mistake lol.

shawnseveras, 2021-01-02:
oh and also it helps to use the original sd card to load the first batch of firmware. but you don't need to add .bin at the end when file name extensions is checked. that's automatic lol. but you will see a 15-20 second delay in the load up screen when it is just blue and then it will flash the logo. that means that it has applied the firmware and should now show up.

> so im hoping this finds everyone well. the solution that has worked for me is really simple but also a duh moment. so the problem isn't with he board,nor the firmware. the problem is windows. when [...]

MatthewKing99, 2021-01-02:
> oh and also it helps to use the original sd card to load the first batch of firmware. but you don't need to add .bin at the end when file name extensions is checked. that's automatic lol. but you [...]

Yep exactly what I did, took me a while to figure it out to lol

tdragovich, 2021-03-06:
For what its worth, I had the same issues. Tried various sdcards and sizes. Solution was as aricut stated above, but because windows 10 thinks that firmware.bin=FIRMWARE.BIN it defaults back to lower case. I had to rename it and then change it to FIRMWARE.BIN

Would have been so much easier if instructions actually used uppercase.

TheRealFlyingCoder, 2021-05-16:
Just an update from me as well, exact same issue, uppercase solution worked 👍 

Rename:

firmware.bin -> firmware1.bin (add any character cause stupid windows equality comparison) -> FIRMWARE.BIN

Padde298, 2021-07-15:
> Just an update from me as well, exact same issue, uppercase solution worked 👍
> 
> Rename:
> 
> firmware.bin -> firmware1.bin (add any character cause stupid windows equality comparison) -> FIRMWARE.BIN

Tried just that but with no success. The firmware fil is renamed to .CUR after I tried. What does that mean?
When I try the flashing, there is a red diode flashing (for a while) on the board. The green diode next to the card reader is lit when the card is inserted.
When powered up, the displays says "No printer connected"...

n5kzw, 2021-07-18:
Ender 3 Pro + BTT skr mini E3 V2 + BTT 3.5 TFT V3.0.  It flashed one upgrade to Marlin 2.0.9.1, and all further attempts to flash changes to Marlin result in the firmware.bin file being renamed to FIRMWARE.CUR but no the new firmware.bin is not flashed.  Running Ubuntu 2.0.4.2 LTS, and using Pliaformio as an IDE.  Firmware.bin files created with platformio build and auto build Marlin.  I would try re-flashing the bootloader, but the only link I found requires changing the SWD pin definitions.  If I could upload a Marlin file to do that, I wouldn't need to re-flash the bootloader to begin with.  Suggestions??

SuperCommands, 2021-08-05:
> Instead of copying th bin file onto the sd card , move it. As I was doing the video to send BIQU it flashed and that was the only thing that I changed. This worked for me . Hope it does for you. Get Outlook for Android<https://aka.ms/ghei36>
> […](#)

Thx, this really helped was having the issue and this worked

hernan43, 2021-08-09:
Just throwing my 2 cents in . I had this exact issue today and it seemed to be solved by my formatting the included SD card as FAT and not FAT32. Once I did that I was golden.

JettScythe, 2021-08-23:
Same kinda issue here. Running Manjaro as a desktop OS so I knew it wasn't some kind of Windows specific behavior. Tried all the different capitalization and file extension configurations - nothing was working. Came down to the bottom of this thread and was _going_ to do as @hernan43 did and format the SD card to FAT but Gparted doesn't offer that. I simply formatted the card to (_get this_) **FAT32 from FAT32** and then it flashed no problem.

hernan43, 2021-08-23:
Did they accidentally format them as FAT31 before the left the factory or something?

bigFin, 2021-09-23:
I needed to format my SD card as FAT32 with a MS-DOS partition table. It failed when I used a GPT.

ncblx, 2021-10-18:
> I don't fu...k believe it!!
> 
> I went to https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/tree/master/firmware right click the file on Chrome and save link as frimware.bin
> 
> It is saving the HTML.
> 
> Anyway the printer didn't load anymore anything. It was creating the FIRMWARE.CUR file and I though why the fuck it is resulting in HTML? Funny way to say I am ok or I am not ok.
> 
> I don't know why I didn't clone the God damn repo!

Click to file and in new window you will fine DOWNLOAD button )) RTFM ;) 
[image]

yael-lorenzo, 2021-10-18:
> > I don't fu...k believe it!!
> > I went to https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/tree/master/firmware right click the file on Chrome and save link as frimware.bin
> > It is saving the HTML.
> > Anyway the printer didn't load anymore anything. It was creating the FIRMWARE.CUR file and I though why the fuck it is resulting in HTML? Funny way to say I am ok or I am not ok.
> > I don't know why I didn't clone the God damn repo!
> 
> [Click to file and in new window you will find DOWNLOAD button](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/blob/master/firmware/firmware.bin) )) RTFM ;)
> 
> [image]

Sadly I know, it was a stupid moment and I kept using the same file :\ thanks for the reply btw.

Enoz211, 2022-01-28:
What you need to do is take a real close look with magnifying glass if u have some at both the top plate and in the slot of the sd card bay and make sure u don't have a bent pin thay are cheap card bays and the way thay tend to end up in relation to the enclosure opening a lot of the time a card gets inserted just slightly off kilter and bends a pint it happened to me lucky I wus able to use a soweing needle and bend/push it back in place wile inserting the card and WA LA after it flashed I shut it down and permanently fixed the pin

slackbits, 2022-02-01:
For those that have the firmware.bin renamed to firmware.cur, but without it looking like things updated issue, one thing that affected me was `EEPROM_SETTINGS`. 😄 If you have `EEPROM_SETTINGS` enabled, then re-flashing the firmware will not change any [settings in the EEPROM](https://marlinfw.org/docs/features/eeprom.html). You have to reset to factory defaults to get the update.

I went through a couple of recompile/re-flash steps and finally changed the `CUSTOM_MACHINE_NAME` parameter to ensure it got re-flashed. (Then I Googled 😄 )

ottonilsson, 2022-04-02:
I had the same problem with the firmware.bin getting renamed to firmware.cur (making me think the firmware update was successful, or at least I used the right file) but without the display connecting to the printer afterwards.
Tried all suggestions above, nothing worked..
Then it went up for me that I was trying to upgrade with the firmware file for the V3.0 board..
I have the TFT35 V3.0 and just assumed the board was V3.0 as well when downloading the firmware file (I actually knew it was V2.0, just didn't think that far).

Using the new "firmware.bin" (all lower case) and waiting for 10-20 sec after turning the printer on, then suddenly the TFT-board beeped and I got a green dot replacing the "No printer attached!" message.

Perhaps someone as stupid as me got helped by this :)

aamcle, 2022-09-27:
Has anybody tried flashing via usb?

jawad092, 2022-11-10:
>  I had the same issue with SKR mini e3 v2, Firmware won't install, I did everything,  format SD card in FAT, FAT32, upper case and lower case firmware.bin …. nothing solved it.
another issue it had was " No Printer Attached " , and i thought it had something to do with firmware, and i changed serial numbers and changed many things because when i would connect it to pc or tft24 it would show this error, but when i remove sd card, it would start working properly  (but I didn't knew firmware wasn't installing, because from 1 month i entered into 3d printing world i know nothing)  .... luckily [Enoz211](https://github.com/Enoz211) mentioned that use magnifying glass and check sd card slot and i did and found two pins were bent, i tried to fix it but they broke and i salvaged and replaced that slot from a broken mp3 player, i inserted sd card, firmware installed and when i tried to connect it to pc, it didn't work and displayed usb not recognized, i installed bigtreetech preconfigured firmware and wow, usb pc started working, display connects normally , 100% working now. all problems fixed.... thank you all and especially Enzo211.  

[image]
[image]
[image]
[image]

AndrewVailcross, 2023-02-08:
I'm using SRK E3 mini with TFT35 V3.0, had the same issue, couldn't flash firmware - my solution was disconnecting TFT35 and switching to the original Ender 3 display.

pepemanboy, 2023-03-20:
I just had this issue, and what fixed it for me was:

SD card must be formatted to have a 4096 allocation size. If the SD card is too small (i.e. 128MB) this might not be possible. If the SD card is too large, it must be partitioned to a smaller size (i.e. 4GB).

MGlaab, 2023-03-29:
I had similar issues. I fixed my problem by using sd card formatter, https://www.sdcard.org/downloads/formatter/ to first format and erase my sd card. Then I copied the firmware to the card and it worked fine in my machine.

celticjaden, 2023-03-31:
I had this same issue today with the skr 3 ez board. I had been flashing firmware just fine but I opened a new window in vss to make some changes and compare them to my previous config/config_adv files. when I compiled the firmware, I used the new window with the newly made config changes to do so. the board wouldn't flash. many attempts and several hours later, I re-compiled the old config settings in the original vss window and the board flashed no problem. so I started comparing again and noticed that I did not change the baudrate from 250000 to 115200 in the new config file/window... changed the rate, compiled and you guessed it, it flashed just fine

MattheweHammond, 2023-06-06:
The solution for me was just renaming the file to "firmware" without the .bin at the end. I am on Windows 11 currently if that helps to anyone else.

djgringoboy2003, 2023-08-21:
I was pulling my hair out with this!
Turned out even after doing a clean in marlin auto build I had to delete a firmware.bin I had saved elsewhere first. I nothiced after this that when I did the buid it was made in the default location.
Worked first time after this!
It was so strange as every flash was savinng as the .cur like it was sucsesfull although it wasnt.

Turns out i just didnt clean up properly before i done a rebuild of the firmware

a22bootleg, 2023-11-11:
I've installed a skr mini e3 v2 in an Ender 3. The board did not come with a micro sd card. I'm trying to install a bl touch and not having any luck. The printer works with the new board.

cj110109, 2024-04-14:
> Did they accidentally format them as FAT31 before the left the factory or something?

Hey thanks for the info, I had the same issue, 3 days now on and off, already ordered 3 new boards until one works, octopus max, kraken, and manta m5p, got so fed up just ordered all the best ones, but the FAT system worked, it flashed and came back .CUR, but now, I can't get in to get serial. This is what I get " ls /dev/serial/by-path/
ls: cannot access '/dev/serial/by-path/': No such file or directory" Same story! So over it, new boards come tommorow, just so frustrating this was a huge build, I turned a Creatbot F430 into a TRUE high-temp Chrisbot, you can check it out on my youtube, I had it working with a E3EZ, but not enough power or outputs for the 6 heat sources, and all the other excessive improvements I did, water cooled everything (stepper motors, all of them, extruder and extruder stepper, nozzle, etc) 4 40 amp ssr's and 2 24v 40 amp mosfets with a 1,200 watt power supply, so this at the end of that build is heartbreaking.

cj110109, 2024-04-14:
My chamber is capable of 200c, bed 225c, and nozzle 550c! True PEEK machine!

[3 more replies]