bigtreetech/BIGTREETECH-SKR-mini-E3 #338: Unable to compile to activate NeoPixels on SKR Mini E3 V2.0 Marlin bug fix 2.0.5.4

jdpg2, 2020-07-23:
### Description

Cant compile FW with NeoPixels enabled

### Steps to reproduce

Latest version of Marlin Bugfix using Marlin supplied config for SKR Mini E3 V2.0 modified for Ender 5

**Expected behavior**
 
A clean FW build.  With #define NEOPIXEL_LED commented out the FW builds and works perfectly.

With it un commented you get this error while building

In file included from Marlin\src\feature/leds/neopixel.h:34:0,
                 from Marlin\src\feature/leds/leds.h:33,
                 from Marlin\src\MarlinCore.cpp:107:
C:\Users\jdpg2\.platformio\lib\Adafruit NeoPixel/Adafruit_NeoPixel.h:361:3: error: 'GPIO_TypeDef' does not name a type
   GPIO_TypeDef *gpioPort;       ///< Output GPIO PORT
   ^~~~~~~~~~~~
*** [.pio\build\STM32F103RC_btt_512K\src\src\MarlinCore.cpp.o] Error 1

### Additional Information

The build libraries uses

Verbose mode can be enabled via `-v, --verbose` option
arm-none-eabi-g++.exe -DTARGET_STM32F1 -D__MARLIN_FIRMWARE__ -DHAVE_SW_SERIAL -DDEBUG_LEVEL=0 -DSS_TIMER=4 -DSTM32_FLASH_SIZE=512 -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-features-dependencies.h
Adding lib_deps for HAS_TRINAMIC_CONFIG...
Adding lib_deps for NEOPIXEL_LED...
Adding lib_deps for HAS_GRAPHICAL_LCD...
Adding src_filter for HAS_GRAPHICAL_LCD...
Adding lib_deps for HAS_LCD_MENU...
Adding src_filter for HAS_LCD_MENU...
Ignoring libs:  ['SPI', 'AsyncTCP', 'SailfishLCD', 'SlowSoftI2CMaster', 'arduinoWebSockets', 'MKS-LittlevGL', 'LiquidTWI2', 'Adafruit MAX31865 library', 'ESP Async WebServer', 'ESP3DLib', 'Arduino-L6470', 'ESP32SSDP', 'TMC26XStepper', 'LiquidCrystal']
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 6.1.1 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32-maple 2.10000.200103 (1.0.0)
 - tool-stm32duino 1.0.2
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SoftwareSerialM> 1.0.0        
|-- <TMCStepper> 0.7.1
|   |-- <SoftwareSerialM> 1.0.0    
|-- <Adafruit NeoPixel> 1.5.0      
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|-- <STM32ADC> 1.0
|-- <EEPROM>
|-- <USBComposite for STM32F1> 0.91
|-- <Wire> 1.0
|-- <FreeRTOS701>
|-- <Servo(STM32F1)> 1.1.2
Building in release mode

CONFIG.H section in question

#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRB  // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  //#define NEOPIXEL_PIN     PC7     // LED driving pin
  //#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
  //#define NEOPIXEL2_PIN    5
  #define NEOPIXEL_PIXELS 3       // Number of LEDs in the strip, larger of 2 strips if 2 neopixel strips are used
  #define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 127  // Initial brightness (0-255)
  #define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup

  // Use a single Neopixel LED for static (background) lighting
  //#define NEOPIXEL_BKGD_LED_INDEX  0               // Index of the LED to use
  //#define NEOPIXEL_BKGD_COLOR { 255, 255, 255, 0 } // R, G, B, W
#endif

jdpg2, 2020-07-23:
Solved this by replacing Adadfruit_NeoPixel@1.5 with https://github.com/bigtreetech/Adafruit_NeoPixel

closed 2020-07-23 by jdpg2

Dierheul, 2020-10-17:
> 
> 
> Solved this by replacing Adadfruit_NeoPixel@1.5 with https://github.com/bigtreetech/Adafruit_NeoPixel

Please can you tell me . Who you copy this Package ?