bigtreetech/BIGTREETECH-SKR-mini-E3 #339: Mini E3 V2 Support Bltouch home and level on Ender5? firmware-bltouch.bin can i have the source?

zonalimitatore, 2020-07-23:
Where i can find the source files of already compilated firmware-bltouch.bin?
This board support bltouch for home and level on Ender5 normal?

tadasi288, 2020-07-24:
I also want this source file.
firmware-bltouch-for-z-homing-ender5.bin

TpyoKnig, 2020-07-24:
I created my own version here: https://github.com/TpyoKnig/SKR-Mini-E3-v2-Ender-5-Pro-Marlin 

Do note that along with the BLTouch (bltouch is plugged into probe port with the pins connected in the Z endstop) I've enabled Z safe homing, and sensorless homing.

zonalimitatore, 2020-07-24:
Yep....but it's for pro

TpyoKnig, 2020-07-24:
Here is mine: https://github.com/TpyoKnig/Marlin-2.0.x-SKR-mini-E3-V2.0 

(Previous was the wrong one). 

The differences between a pro and non-pro aren't hardware size (other than maybe the lead screw on the updated models) So this should work the same. :)

MadMathWizard, 2020-08-10:
@TpyoKnig is this for 2.0.5? I tried to copy your configuration.h and configuration_adv.h into mine and got version errors.
I'm a novice and sorry for the trivial question.

TpyoKnig, 2020-08-10:
@MadMathWizard no this is for 2.0-bugfix. There is a newer version (2.0.6) that is out but I've not yet updated the config to be compatible with it (2.0.6) as 2.0.5 does not have support for the SKR Mini E3 V2.

MadMathWizard, 2020-08-10:
Maybe that is my problem then. I'm trying to use 2.0.6. I'll try using bugfix.
Thanks!

TpyoKnig, 2020-08-10:
@MadMathWizard if you clone my entire repo, that's the specific version of 2.0-bugfix that you need, although I do need to update it to 2.0.6 since there are some other additional enhancements in 2.0.6. 

Downloading the current 2.0-bugfix may not work as IIRC they've renamed the board as the 2.0-bugfix that was used was one that BTT added support themselves before it was adopted/adapted upstream in what became 2.0.6.

MadMathWizard, 2020-08-10:
Great, I'll do that. Thank you.

TpyoKnig, 2020-08-10:
@MadMathWizard I just pushed a commit to update my entire repo to 2.0.6. I've not fully tested it as I have my Ender5 apart ATM but it should work as I edited the configs to match and was able to compile successfully.

MadMathWizard, 2020-08-10:
I’ll try that later tonight. I tried to compile the other one but I was getting a directory error. Definitely a new user error. Thanks again.

dolphinituk, 2020-08-11:
Think I might give your firmware a go tomorrow. Not tried sensorless homing yet but may solve an issue where my BLTouch every so often fails probing and crashes into bed :-(

MadMathWizard, 2020-08-11:
I gave it a go late last night. So far so good.