bigtreetech/BIGTREETECH-SKR-mini-E3 #340: SKR-mini-E3 V2.0 Sound problem TMC motor

Burockk, 2020-07-24:
Hello,

I have been using this mainboard for 1 month and it is not as quite as creality silent board somehow I can hear some sound on x and Y motors when it is printing Ofcourse it is not as loud as stock board but since it is using TMC 2209 drivers it should not be louder than default silent board I already disabled   //#define HYBRID_THRESHOLD   but is it still not perfect I configured TMC drivers - Driver current X Y Z to 100 it was 1000 and it was louder than now but is there something else I can do to fix this problem?

Ty,

Whiskaya, 2020-07-26:
Agree, the SKR sounds a bit louder.
But then again, the Creality v1.1.5 uses Vref to adjust the 2208s, whereas the SKR Mini adjusts current directly from the firmware to the 2209s.
As these adjusting systems are not comparable, maybe the values in the SKR are really "much higher" than the ones from the Creality, and you just haven't realized.
Try lowering the current to the drivers without losing steps, maybe that'll get it quieter.

vicewize, 2020-07-28:
Plus the 2208s produce less torque overall compared to 2209s.

Burockk, 2020-07-31:
After Watching this video https://www.youtube.com/watch?v=dOJbSrWVu_Q&feature=youtu.be
I edited my settings according to tmc 2209 driver and there is a huge difference it is now as quite as creality silent board.

Liamnerf, 2020-08-01:
I can confirm this issue as well. just switched from the v1.2 to the V2.0 and it's way louder. wasn't expecting a downgrade, very annoying...

Burockk, 2020-08-01:
> I can confirm this issue as well. just switched from the v1.2 to the V2.0 and it's way louder. wasn't expecting a downgrade, very annoying...

Watch the video I wrote and change those values it helps a lot.

Liamnerf, 2020-08-01:
> > I can confirm this issue as well. just switched from the v1.2 to the V2.0 and it's way louder. wasn't expecting a downgrade, very annoying...
> 
> Watch the video I wrote and change those values it helps a lot.

Thanks, it’s all good now though, turns out I had my homing current set way too high. 6500 instead of 650 😳