bigtreetech/BIGTREETECH-SKR-mini-E3 #343: Cannot compile firmware

tk9851, 2020-07-30:
Hello all,

I am trying to compile Marlin based on the "Marlin-2.0.x-SKR-mini-E3-V2.0" folder, without any changes, and I am receiving the following error. Any help is appreciated!
[image]

redgod, 2020-08-02:
yo tampoco puedo skr mini 1.2 

[image]

rmacewen1, 2020-08-03:
I was struggling myself.  It's all new environment to me.  I found that what solved on problem for me that came up, after restarting this project fresh was to make sure I scroll down on left pane, under Platformio, and select the ENVIRONMENT shown as:
env:
[image]

I'd scroll upward in your notifications and see if there is a hint as to the error.  Actually it looks like you had a correct environment selected.  :/

redgod, 2020-08-04:
sigo sin poder compilar :( 

[image]

psycor, 2020-08-04:
Hello,
it would be very helpful if i could download the firmware not only as a ready to use bin-file in different versions, but also as a marlin zipfile. with all setup files, so i only have to activate the filamentsensor and everything works.
My problem is that I can't find all settings that are available in the finished firmware to configure it with Marlin completely by myself. e.g. I can't find the setting for the x-y offset for BLTOUCH.

NameOfTheDragon, 2020-08-04:
@psycor There are really only 2 files you need to look at: config.h and config_adv.h, everything you would ever need to change is in those 2 files. Just go through them line by line until you find what you need. You can download the Marlin source code as a zip file right from GitHub in the Marlin repository.

psycor, 2020-08-04:
@NameOfTheDragon I do it more than 3 times but i can´t find it. Maybe i find it in a while, but today i break with it. Much thanks