bigtreetech/BIGTREETECH-SKR-mini-E3 #345: [BUG] (short description)

krisald444, 2020-07-31:
### Description

Receive ERR Maxtemp on bootup, likely shorted something while cleaning with brass wire brush while hotend was on and now only receive err maxtemp on boot, I've tested thermistor and heat cartridge both are still good, only thing that I was able to do to get past ERR maxtemp was to disable thermal runaway and raised hotend max temp to 3000 deg but with nothing connected to board and only powered up by usb it now shows hotend 300 deg celsius, I'm assuming something on the board is bad now but dunno what,  is it possible to change thermistor to a different pin with firmware or test/change resistor, mosfet or something else that might be blown or is the board a paper weight now

### Steps to reproduce

1. [Tested thermistor/ heat cartridge]
2. [Replaced thermistor/heat cartridge]
3. [Recompiled firmware]
**Expected behavior**
fully boot with correct temperatures

**Actual behavior**
after changing hotend maxtemp in firmware to 3000°c it fully boots but shows hotend is 300°c on bootup 

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

closed 2020-08-05 by krisald444

Chanvgap, 2020-09-30:
Did you ever get an answer to this?   I believe i have the exact issue.  Was hoping there was a solution other than ordering a new board.

krisald444, 2020-10-07:
Its possibly the 35v 10uf capacitor that blew on TH0 but haven't actually
verified, I'm just going by the schematic someone found

wikolii, 2021-05-10:
Hi, were you able to find a solution to this? i'm facing the same issue and i was able to get past the "max temp bed" error by changing the max temp. it's reading 159ºC but i'm looking for the underlying issue. Maybe it's a component that can be changed.

Thanks!

wikolii, 2021-05-10:
If you switched a faulty component could you point which one? so i can test it? thanks!

TheKurgan, 2021-07-27:
I have had the same issue, and am searching for a DIY fix... I am very good at soldering small components, but I do not know how to find which component has failed.

dderg, 2023-02-09:
mine shows 300 degrees for bed temperature even with thermistor disconnected