bigtreetech/BIGTREETECH-SKR-mini-E3 #346: [BUG] (short description)

jphansendk, 2020-08-01:
### Description
I can not save to EEPROM from marlin or terminal. When I flash new firmware (bugfix 30/7-20) I do M502, M500 and get: echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Settings Stored (4 bytes; crc 7439)
Error:EEPROM datasize error.
Error:EEPROM datasize error.

When I run m504 I get:

SENDING:M504
echo:EEPROM version mismatch (EEPROM=? Marlin=V81)

In configuration_adv.h I have tried with: #define I2C_EEPROM
#define MARLIN_EEPROM_SIZE (4*1024)
I get same error.

Only this makes EEPROM work:
#define FLASH_EEPROM_EMULATION
#define EEPROM_PAGE_SIZE     (0x800U)
#define EEPROM_START_ADDRESS (0x8000000UL + (STM32_FLASH_SIZE) * 1024UL - (EEPROM_PAGE_SIZE) * 2UL)
#define MARLIN_EEPROM_SIZE    EEPROM_PAGE_SIZE

### Steps to reproduce

1. Initialize EEPROM 
2. Flash new bugfix firmware from Marlin github and BBT github
3. Run M502 and M500
4. Reboot

**Expected behavior**
Save settings to internal EEPROM  

**Actual behavior**
Will not save settings to internal EEPROM

### Additional Information
`Configuration.h`: https://pastebin.com/RuC7jiph
Configuration_adv.h`: https://pastebin.com/zxi8NYXu

Glychd, 2020-08-01:
I'm having the same issue, with the same error code. I've tried flashing several different versions of marlin, and several different configs. Nothing has had any effect. This is on the SKR mini e3 v2, installed in an ender 3 pro.

jphansendk, 2020-08-01:
> I'm having the same issue, with the same error code. I've tried flashing several different versions of marlin, and several different configs. Nothing has had any effect. This is on the SKR mini e3 v2, installed in an ender 3 pro. 

Ender 3 in my case.

SybexX, 2021-01-04:
I'm having the same issue, with the same error. By changing the Marlin-2.0.x\Marlin\src\HAL\shared\eeprom_if_i2c.cpp file
I was able to eliminate the error.

/**********************************************************************************************/
#include "../../inc/MarlinConfig.h"

#if ENABLED(I2C_EEPROM)

#include "eeprom_if.h"
#include <Wire.h>

void eeprom_init() { Wire.begin(); }

#if ENABLED(USE_SHARED_EEPROM)

#ifndef EEPROM_WRITE_DELAY
  #define EEPROM_WRITE_DELAY    5
#endif

#ifndef EEPROM_DEVICE_ADDRESS
  #define EEPROM_DEVICE_ADDRESS  0x50
#endif

//static constexpr uint8_t eeprom_device_address = I2C_ADDRESS(EEPROM_DEVICE_ADDRESS);

// ------------------------
// Public functions
// ------------------------

void eeprom_write_byte(uint8_t *pos, unsigned char value) 
  {
  const unsigned eeprom_address = (unsigned)pos;

  Wire.beginTransmission(EEPROM_DEVICE_ADDRESS);
  Wire.write(int(eeprom_address >> 8));   // MSB
  Wire.write(int(eeprom_address & 0xFF)); // LSB
  Wire.write(value);
  Wire.endTransmission();

  // wait for write cycle to complete
  // this could be done more efficiently with "acknowledge polling"
  delay(EEPROM_WRITE_DELAY);
  }

uint8_t eeprom_read_byte(uint8_t *pos) 
  {
  const unsigned eeprom_address = (unsigned)pos;

  Wire.beginTransmission(EEPROM_DEVICE_ADDRESS);
  Wire.write(int(eeprom_address >> 8));   // MSB
  Wire.write(int(eeprom_address & 0xFF)); // LSB
  Wire.endTransmission();
  Wire.requestFrom(EEPROM_DEVICE_ADDRESS, 1);
  return Wire.available() ? Wire.read() : 0xFF;
  }

#endif // USE_SHARED_EEPROM
#endif // I2C_EEPROM
/**********************************************************************************************/

my configuration of the EEprom, entered in the pins_BTT_SKR_V1_4_TURBO.h
-------------------------------------------------------------------------------
#define I2C_EEPROM

#ifdef E2END
  #undef E2END
#endif 

#define E2END	0xFFFF		// AT24C512
//#define E2END	0x7FFF		// AT24C256

BlueHornet, 2021-01-29:
@SybexX 
Interesting I have the exact same issue with my Ender 5 Plus & BTT SKR 1.4 Turbo / BTT TFT35 V3 E3, 
I applied your fix but the error still persists. My EEROM is 256 so commented out the 512 line & commented 256 line

TheGunPsy, 2021-06-12:
> #define FLASH_EEPROM_EMULATION
> #define EEPROM_PAGE_SIZE (0x800U)
> #define EEPROM_START_ADDRESS (0x8000000UL + (STM32_FLASH_SIZE) * 1024UL - (EEPROM_PAGE_SIZE) * 2UL)
> #define MARLIN_EEPROM_SIZE EEPROM_PAGE_SIZE

I noticed that you have "STM32_FLASH_SIZE". is that the chip on the board. I know my old SKR E3 Mini used a similiar board name. I am running the SKR E3 Turbo that uses the LPC1769. Would this string of code need to be different?

Barkmadly, 2021-08-15:
@SybexX 
Worked for me.  Thanks!  I'd been trying for days to fix this until I copied your code.
I initially @ucked my eeprom by writing corrupt firmware onto it.