bigtreetech/BIGTREETECH-SKR-mini-E3 #349: [BUG] (short description)

Swift-Tester, 2020-08-03:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/BTT%20SKR%20MINI%20E3%20V2.0SCHpdf.PDF shows no sign of pullup/current_limiting schematics both at SERVOS and PROBE pins, connected directly to PA1 and PC14 I/O of MCU, which are NOT 5V tolerant.

Please confirm this being:
a) typo in schematics in the PDF (linked above) or
b) regression in real PCB, comparing to the previous v1.2 board.

Do we need extra circuits to connect clones (3DTouch) with 5V levels to 5-pin Z-PROBE socket of BTT SKR MINI E3 V2.0 ?

jphansendk, 2020-08-10:
As far as I read the z-endstop pin is 3.3v but I can't tell if this change when you set the 5v jumper on the board.. don't know how to measure that pin with a multimeter.. not an electronics guy.

Swift-Tester, 2020-08-11:
The pins I'm talking about are from Z-PROBE socket, not from Z-STOP.

jphansendk, 2020-08-12:
Aah sorry.. I guess the board has some power problem in both then.. I have probe issues when probe is connected to z-stop as well.. the probe keeps dropping in print.. 😩