bigtreetech/BIGTREETECH-SKR-mini-E3 #350: Auto home not working - 5v power module the problem

bucefalo2, 2020-08-03:
### Description
Ender 3 is NOT going home ! i found the problem and i explain it here in this new post 
PROBLEM:
Ender 3 suddenly would not go HOME ( nozzle in home position ) if i would click on AUTOHOME
luckily i checked if cable connections were good and i removed mobo fan connection and AUTO HOME worked again but as soon as i'f plug it back than it would not work anymore !
I had installed BIGTREETECH DCDC5V V1.0 Power Module 5V (for neopixel module) but never used it so i thought let's remove it and plug back mobo fan and YES it works.. AUTOHOME works ! so probably the 5v module is the problem?? how can it be ?  

### Steps to reproduce

1. [First Step] i turn on ender 3 .. and i decide to send nozzle to home
2. [Second Step] 
nozzle would move in X aand Z position but not Y 
3. [and so on...]

**Expected behavior**
nozzle should have gone to HOME position in all axis 

**Actual behavior**
nozzle went home in Z and X axis but Y not going

### Additional Information
ender 3 has been working perfectly until yesterday ! than decided to no go home (nozzle in home position )
today casually i found out that disconnecting mobo fan and later 5v power module helped!
now mobo fan is connected and working and AUTO HOME working !

bucefalo2, 2020-08-04:
EDIT : 
problem appeard again later... i don't know what is causing it ! but wasn't the 5v module !

actually when i could finally connect it to my pc with usb cable via pronterface i could send it home and store the setting !
now it's going home , all axis take nozzle at home  so probably if i plug back the 5v module it'll still work fine !
not sure how this problem is triggered

Pmpnak, 2020-08-07:
I am brand new to 3d Prining. I purchased the Ender 3 pro and added the skr mini e3 2.0 and tft 35 e3 3.0 during construction. I turned it on per videos and went to use the home and I’m getting the her bed y running into the stops and the x as well. Will not perform the y up as well. You said you hooked  the printer up to your pc and than what? I don’t understand how to get a firmware update to see if that fixes the problem or what the steps you took mean. Would it be possible to explain this better please? Thank you b

bucefalo2, 2020-08-12:
i really wish i could help .. but the problem came and disappeard " almost " by itself ! 
somehow when i connected 3d printer through usb on my pc and sent HOME input with pronterface it recognized " home " the  correct one and since then is taking the nozzle at " home " ! i hope you can solve it too