bigtreetech/BIGTREETECH-SKR-mini-E3 #352: collect2.exe: error: ld returned 1 exit status -  SKR Mini E3 2.0 / Ender 3 Pro

ventoetc, 2020-08-06:
### Description
Firmware will not complete compilation ending with a "collect2.exe: error: ld returned 1 exit status

When compiling, the process almost completes but fails during creation of the firmware file. 

Linking .pio\build\STM32F103RC_btt_512K\firmware.elf
.pio/build/STM32F103RC_btt_512K/src/src/HAL/shared/eeprom_if_i2c.cpp.o: In function `eeprom_write_byte(unsigned char*, unsigned char)':
C:\Marlinbuild/Marlin\src\HAL\shared/eeprom_if_i2c.cpp:52: multiple definition of `eeprom_write_byte(unsigned char*, unsigned char)'
.pio/build/STM32F103RC_btt_512K/src/src/HAL/shared/eeprom_i2c.cpp.o:C:\Marlinbuild/Marlin\src\HAL\shared/eeprom_i2c.cpp:52: first defined here
.pio/build/STM32F103RC_btt_512K/src/src/HAL/shared/eeprom_if_i2c.cpp.o: In function `eeprom_read_byte(unsigned char*)':
C:\Marlinbuild/Marlin\src\HAL\shared/eeprom_if_i2c.cpp:66: multiple definition of `eeprom_read_byte(unsigned char*)'
.pio/build/STM32F103RC_btt_512K/src/src/HAL/shared/eeprom_i2c.cpp.o:C:\Marlinbuild/Marlin\src\HAL\shared/eeprom_i2c.cpp:97: first defined here
.pio/build/STM32F103RC_btt_512K/src/src/HAL/shared/persistent_store_api.cpp.o:C:\Marlinbuild/Marlin\src\HAL\shared/persistent_store_api.h:53: multiple definition of `persistentStore'
.pio/build/STM32F103RC_btt_512K/src/src/HAL/shared/eeprom_api.cpp.o:C:\Marlinbuild/Marlin\src\HAL\shared/eeprom_api.h:71: first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\STM32F103RC_btt_512K\firmware.elf] Error 1
==============================================================

### Steps to reproduce

1. Customize configuration.h, and configuration_adv.h
2. Compile using correct build environment

**Expected behavior**
I expect the process to complete. 

**Actual behavior**
The process ends during the firmware generation process at the end. 

### Additional Information
[STM32F103RC_error_eeprom.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5035072/STM32F103RC_error_eeprom.txt)

[Configuration_adv.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5035083/Configuration_adv.txt)
[Configuration.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5035084/Configuration.txt)

ventoetc, 2020-08-06:
I copied each configuration line to the bug fix marlin and it worked for me.

closed 2020-08-13 by ventoetc

jphansendk, 2020-08-14:
Did you try to delete your .pio folder at build again?

Also see here: https://github.com/MarlinFirmware/Marlin/issues/16463