bigtreetech/BIGTREETECH-SKR-mini-E3 #353: [BUG] no screw hole in SKR mini E3 v2.0 CAD model

riaqn, 2020-08-06:
### Description

The 3d model (.step) file provided for SKR mini E3 v2.0 does not contain screw holes.
On the other hand, the [size pdf file](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/BTT%20SKR%20MINI%20E3%20V2.0-SIZE.pdf)  contains the distance between holes, but nothing about the hole size, nor the distance from the hole to the edges.

latonita, 2020-09-01:
+1. downloaded 3d model just to insert it to fusion assembly and make holder.... no screw holes... 👎 👎 👎

riaqn, 2020-09-04:
@latonita Could you try the file I provided above and let us know if it's precise?

kifantidis, 2020-10-08:
Hello @riaqn . 
I have downloaded your file but according to [size description file of skr mini e3 v2.0](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/BTT%20SKR%20MINI%20E3%20V2.0-SIZE.pdf) provided in BTT github you have not the exact same dimensions. But it's looking kinda decent. I'm doing the holes atm just like the BTT file and I'll upload a repo later.

riaqn, 2020-10-08:
@kifantidis As I pointed out above, NOTE: the size pdf file provided by this repo seems to be not complete and wrong in several measurements.

kifantidis, 2020-10-08:
@riaqn thank you for the quick response. I've seen your comment above. I didn't read it properly before.

Regards, Kostas.

ianfights, 2023-12-29:
There are also no screw holes in the V3 step file as well, however I managed to find [this](https://grabcad.com/library/btt-e3-mini-v3-1) which has the correct screw holes and the V3's heatsink