bigtreetech/BIGTREETECH-SKR-mini-E3 #354: Compilation terminated

Caspar311, 2020-08-07:
**Good morning everybody, I've been trying to compile the firmware for a whole day. Always the same problem. With platform IO under Atom or VSC.
I tried to compile version 1.0 and 1.2 and 2.0, always the same error.**

.
.
.
.
Compiling .pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\eeprom_flash.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\eeprom_if_iic.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\eeprom_sdcard.cpp.o
Compiling .pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\eeprom_wired.cpp.o
In file included from Marlin\src\HAL\STM32F1\dogm\../../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\STM32F1\dogm\../../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\STM32F1\dogm\../../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp:29:
Marlin\src\HAL\STM32F1\dogm\../../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\STM32F1\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\STM32F1\eeprom_bl24cxx.cpp:28:
Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\eeprom_bl24cxx.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\STM32F1\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\STM32F1\eeprom_if_iic.cpp:28:
Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
ompilation terminated.
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\eeprom_if_iic.cpp.o] Error 1
========================= [FAILED] Took 39.97 seconds =========================
 Environment               Status    Duration
------------------------  --------  ------------
.
.
.
STM32F103RC               IGNORED
STM32F103RC_fysetc        IGNORED
STM32F103RC_btt           IGNORED
STM32F103RC_btt_USB       IGNORED
STM32F103RC_btt_512K      FAILED    00:00:39.971
STM32F103RC_btt_512K_USB  IGNORED
STM32F103RE               IGNORED
STM32F103RE_btt           IGNORED
STM32F103RE_btt_USB       IGNORED
.
.
.

Does somebody has any idea?

Many thanks!
Florian

jphansendk, 2020-08-07:
Try to delete the .pio folder restart vsc and build. See if it makes a difference?

Caspar311, 2020-08-07:
> Try to delete the .pio folder restart vsc and build. See if it makes a difference?

Hi, sorry unfortunately makes no difference.

But as described [here 302](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/302), it has already fixed a bug.

 - Find and replace the line "platform = ststm32" with "platform = ststm32@<6.2.0"

But now comes the next problem ... :
 
- " Please select only one of SD CARD, FLASH, or SRAM EEPROM EMULATION "

I'm going crazy :)

closed 2020-08-07 by Caspar311

Stormsys, 2021-05-09:
@Caspar311 - how did you solve this - i dont seem to be able to fix it

Caspar311, 2021-05-09:
@Stormsys
Hey Diogo,
I deleted the "Build" folder... After this i've no more problems.