bigtreetech/BIGTREETECH-SKR-mini-E3 #356: Ender 5 Pro-SKR MINI E3 V2.0 Z-Azis Flipped

Avery422, 2020-08-09:
I replaced the silent board on my ender 5 with a SKR MINI E3 V2.0 and a TFT 35 V3.0. I installed the firmware from  (https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V2.0/firmware-ender5.bin) the axis goes the wrong way until crashing into the control box. The x and y are okay.

### Steps to reproduce

1. Install the firmware from https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V2.0/firmware-ender5.bin

**Expected behavior**
As part of auto-homing the bed is supposed to go down a bit then up until the Z stop switch activates

**Actual behavior**
3. Bed goes down until it hits the control box
### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

MadMathWizard, 2020-08-10:
I will add, that I have this exact same issue.

Avery422, 2020-08-12:
Here is what I use. Works for my Ender 5.  I also have linear advance enabled but not active (set to 0) by default
[firmware-srmod (1).zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5062628/firmware-srmod.1.zip)

Birdy74, 2020-08-29:
> Here is what I use. Works for my Ender 5. I also have linear advance enabled but not active (set to 0) by default
> [firmware-srmod (1).zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5062628/firmware-srmod.1.zip)

Does not work for me. No BLTouch option in menu either now...

DrDMo, 2020-09-14:
Avery422 - got the code for this bin? I've been struggling with it for weeks and yours is the only one I can get that works, but I would like to add my BLTouch to it! Thanks!!

DrDMo, 2020-09-14:
> Here is what I use. Works for my Ender 5. I also have linear advance enabled but not active (set to 0) by default
> [firmware-srmod (1).zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5062628/firmware-srmod.1.zip)

Avery422 - May I please have the Configuration file(s) for this bin? I've been struggling with this for weeks and your bin file is the only one I can get to work, but I would like to add my BLTouch to it! Thanks!!

Avery422, 2020-09-14:
Hey I have a new one that has a bl touch skr mini e3 v2 and tft 35

Avery422, 2020-09-14:
Would that work

DrDMo, 2020-09-14:
> […](#)

Yes, I suppose. I was hoping to get the Configuration.h and other files so I could make my own changes on some other stuff, but I'll take what I can get, thanks.

DrDMo, 2020-09-14:
Just for clarity: I'm using an Ender 5 Pro and the example delivered with the latest Marlin does not work at all - it's a mess, actually. Do you think I should switch over to the Ender 5 code and give it a go? That's what you're using, right?

Avery422, 2020-09-14:
That was not my own firmware. I found it somewhere and was just saying that was what I was using. I have one that uses a bl touch, with skr mini e3 v2.0, and tft 35 e3 v3.0. 
[Firmware Files.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5215813/Firmware.Files.zip)

Avery422, 2020-09-14:
This is what I use. I dont have the config.h files anymore but I can see if I can find them. I use this with an ender 5 pro with a bl touch, with skr mini e3 v2.0, and tft 35 e3 v3.0. I would stick with the code in the comment above and see if that works on your machine.

Avery422, 2020-09-14:
If you need more clarifications with the steps of uploading the new firmware or any questions, I will try to get back to you as soon as possible.

DrDMo, 2020-09-14:
> If you need more clarifications with the steps of uploading the new firmware or any questions, I will try to get back to you as soon as possible.

Ah, copy that. Thank you so much for this - I'm back in business!

Avery422, 2020-09-14:
Good to hear, Glad I could Help.

StellasDad, 2021-08-02:
I read the comments, but I'm a tad confused (newbie). I'm also looking for E5Pro firmware for the SKR E3 Mini v2 with BL Touch. Can someone please guide me to the right one?