bigtreetech/BIGTREETECH-SKR-mini-E3 #359: BLtouch Probing Error

Ed-Savill, 2020-08-10:
I got stuck trying to add a BLtouch & NeoPixel setup to my Ender 3 with the SKR v2 mainboard.

The works fine with the BTT pre-configured firmware (firmware-bltouch.bin).

I tried to add the BLtouch settings, along side NeoPixel settings. The NeoPixel appeared to work, but the BLtouch wasn't playing ball. I kept getting a probing error.

I have given up now, after playing with the config for over 6 hours.

I would love to get NeoPixel working alongside the BLtouch if anyone is willing to help me, or send me over a working config please!

Ideally, if I could have the config that firmware-bltouch.bin was compiled from, that would be great, so I can add my NeoPixel settings.

aquacidex9, 2020-08-15:
i too would love to have it, i could not get the bltouch to work correctly but the firmware-bltouch.bin works great, i would like to be able to turn on linear advance though

cmmx34, 2020-08-16:
I just installed my BL-Touch and I'm getting the same probing error. ABL is not behaving correctly with Firmware-BLTouch.bin installed. ABL should cause the hot end / BL-Touch to move over the bed, the probe extend, then the Z axis lowers until the probe touches the bed. Instead, the nozzle moves over the bed, the probe extends, but the Z axis does not lower to the bed. The probe never touches the bed, so it generates an error. How do we correct this?

codedmind, 2020-08-27:
Same issue, bltouch won't go down and won't prove

aquacidex9, 2020-08-27:
I figured out you have to disable both make probe same as stop and the specify probe pin it worked after that

codedmind, 2020-08-27:
@aquacidex9 sorry, can't understant what you say... can you be more clear please?

cmmx34, 2020-08-27:
I connected mine substituting the z-axis connector with the probe
connector. I then updated the firmware with the corresponding version from
the Github site and input a large z offset of -3.40. The first time I
printed, though, I had to do some manual bed adjustments on the left side.
It's been printing well ever since.

The problem I have now, and Creality support hasn't replied to my inquiry
about it, is how do I know if the probe is really doing anything - or is
the bed just properly leveled.

Also, after running ABL, where are the adjustment settings stored? Can I
review them?

NameOfTheDragon, 2020-08-27:
@cmmx34 the mesh is stored in the EEPROM provided you save the settings. You can generally view the current mesh data using `G29 T0`.

A better way is to use OctoPrint with the Bed Visualizer plugin installed.

cmmx34, 2020-08-27:
Thanks, Tim. I'll take a look.

I've been deferring the purchase of Octoprint. After buying the printer,
the Bigtreetech board/lcd combo and the BL-Touch, plus several reels of
PLA, I need to conserve funds a bit. It looks quite good, though.

NameOfTheDragon, 2020-08-27:
OctoPrint itself is free – but you need a Raspberry Pi (or other Linux based device) to run it on. A Pi 3B+ is more than adequate.
Best regards,
Tim Long

cmmx34, 2020-08-27:
Thanks, I plan to look into it.

gurt-il, 2020-10-22:
I also have the same problem.
I've managed to fix it by setting the Z Offset upfront to some value (I've started with -2.50) and then it's worked and I was able to test for the correct Z Offset.
The problem is when the offset is too big, the probing failed.

Just remember to add the offset you see on the screen to the offset you already set.