bigtreetech/BIGTREETECH-SKR-mini-E3 #360: SKR Mini E3 V2 Thermal Runaway E1

ambrosv72, 2020-08-13:
For no reason i started to get "Thermal Runaway E1" error as soon as i turn my Ender 3 Pro on.  I replaced thermistor and still get error.  So I disconnected all plugs and wires to SKR mini and just left the power source and the display plugged in.  Power the board up and I still am getting the error.  Did the board just fail?  It was working fine for about 2 months.  This just happen all of a sudden.  Does anyone have any idea what would cause this?  And/Or resolution?  Thanks.

Drewster65, 2020-08-27:
same problem here on the 1.2 vers.  working fine and threw this error, board wont recover and its only 6 weeks old. this is the second board to fail this way in less than 3 months. really not happy with this product when it is so unpredictable.

LearnIIBurn, 2020-09-04:
Happened to my first mini-e3 v2.0. Never did resolve it no matter what I tried. Bought a second board and installed it today and now I get a whole new boot issue, the blank screen and inability to flash firmware. What a dumpster fire product this is.

ptorres98, 2020-11-15:
I having thermal runaway probs with the E1... wen starts the cooling fan... lack of power?

p6002, 2021-11-12:
Same error here. New board showing error when heating nozzle for ABS print. Ender 3. Skr mini v2. 
Any ideas how to fix it?

wdrovere, 2021-11-13:
in my case, the terminal where we connected the thermistor short-circuited... i contacted btt, they returned me a coupon... i tested it with multimeter and beepppp

tomcarrolls, 2021-11-13:
They returned you a 'coupon'? You tested the coupon with a multimeter? What?