bigtreetech/BIGTREETECH-SKR-mini-E3 #361: marlin skr e3 mini v2.0 won't compile

DRCUE2004, 2020-08-15:
I have tried all the solutions that I have read .. and none have worked
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3

It always turns out the same mistake.
**STM32F103RC_btt_512K      FAILED**

https://ibb.co/brSCWhH

I have read this question and the similar ones
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/334

Thanks ;)

JonT79, 2020-08-15:
I struggled with this issue just yesterday. I got the same error downloading Marlin files from SKR V2.0 on GitHub doing a google search. Instead I downloaded the latest version from Marlinfw.org, then download the Marlin config\examples from github and copy paste 4 files from Creality\Ender3\SKR...V2.0 (configuration.h, configuration_adv.h... etc). Built fine for me afterwards. Make sure to change  “default_envs = STM32F103RC_btt_512K” In platform.io. Hope this helps.

DRCUE2004, 2020-08-15:
Ok thanks for the information.

But the compilation fails at the same point

JonT79, 2020-08-15:
Did you download from the link you provided? https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3 
When I used that download I got that same compile error. Did you instead try downloading from?https://marlinfw.org/meta/download/ 
that worked for me but you will have to get the example config for your board.

DRCUE2004, 2020-08-15:
to the third .. it seems that it does not give an error 👍 

I have used these 4 files
https://github.com/MarlinFirmware/Configurations/tree/release-2.0.6/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20Mini%20E3%202.0

pasted in the Marlin folder of:
Latest releaseSupports AVR and ARMArduino and PlatformIO | 2.0.6 | 2.0.x.zip
https://marlinfw.org/meta/download/

thanks for everything

I can now start testing :)

JonT79, 2020-08-15:
Glad it worked out! Like I said I spent many hours yesterday with the same issue. Btw this guide is very helpful. 
https://www.reddit.com/r/ender3/comments/h8y1ia/marlin_20x_guide_skr_mini_e3_v20_ender_3/?utm_source=xpromo&utm_medium=amp&utm_name=amp_comment_iterations&utm_term=active_search&utm_content=post_body

DRCUE2004, 2020-08-15:
ok, thanks for the guide

I will read it carefully   :)

Typeopymp, 2020-08-16:
I'm having the same problem as OP,  I tried the above and I get a ton of errors then the same Failed message at the bottom:

[image]

[image]

Typeopymp, 2020-08-16:
Well I found what I did wrong but I have a new subtle error:

[image]

DRCUE2004, 2020-08-16:
I have followed these steps to compile it :)
https://youtu.be/mtCz_-2zvZo?t=373

JonT79, 2020-08-17:
DRUE2004: One thing I have found. If you follow the changes in the guide I linked I would not enable SQUARE_WAVE_STEPPING since it doesn’t work well with the TMC2209 drivers. I had worse prints after enabling this. 

Typeopymp: the only thing I can say is make sure you get a version of Marlin that compiles right off the bat (without any changes) then check all of your changes very carefully for typing mistakes. The error messages will usually point you to where the mistakes may be. If it compiles the firmware.bin file should be in “.pio/build/STM32F103RC_btt_512k

DRCUE2004, 2020-08-17:
Vale ... No habilitaré SQUARE_WAVE_STEPPING

thanks for the warning ;)

JonT79, 2020-08-18:
FYI I made a post on thingiverse trying to resolve having worse prints with the new board. I think I came up with some pretty good findings that will make your prints much better!

https://www.thingiverse.com/groups/ender3/forums/general/topic:47052

Photo: left to right: Original Ender-3 board, SKR V2.0 board pre-loaded firmware, SKR V2.0 modified firmware

[image]