bigtreetech/BIGTREETECH-SKR-mini-E3 #362: Cant Build Firmware for SKR mini v2.0

JoshuaTheBaum, 2020-08-15:
Hi
I installed a BL Touch (from Antclabs) and the Micro Swiss Direct Drive Extruder Setup. Now i need to change the #define X_MIN_POS and #define Y_MIN_POS to #define X_MIN_POS -5 and #define Y_MIN_POS -10 . 
I downloaded the Firmware here and changed the parameters. Then i tried to compile it with Platform.IO in VSCode. But i always got the STM32F103RC_btt_512K      FAILED  Error. I looked here for fixes but found nothing working for me. 
Im kinda new to the Marlin building so i hope its just beginner mistake :D

Heres the Error Message

 Executing task in folder Marlin-2.0.x-SKR-mini-E3-V2.0: C:\Users\joshu\.platformio\penv\Scripts\platformio.exe run <

Processing STM32F103RC_btt_512K (platform: ststm32; board: genericSTM32F103RC; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 8.0.0 > STM32F103RC (48k RAM. 256k Flash)    
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 4.10900.200602 (1.9.0)
 - framework-cmsis 2.50501.200527 (5.5.1)
 - tool-stm32duino 1.0.2 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing Adafruit_NeoPixel
UserSideException: Please install Git client from https://git-scm.com/downloads:
  File "C:\Users\joshu\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 168:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\joshu\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\joshu\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\joshu\.platformio\platforms\ststm32\builder\main.py", line 99:
    target_elf = env.BuildProgram()
  File "C:\Users\joshu\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\joshu\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\joshu\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\joshu\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 131:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\joshu\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\joshu\.platformio\penv\Lib\site-packages\platformio\builder\tools\piolib.py", line 1036:
    project.install_dependencies()
  File "C:\Users\joshu\.platformio\penv\Lib\site-packages\platformio\builder\tools\piolib.py", line 868:
    lm.install(uri)
  File "c:\users\joshu\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 275:
    pkg_dir = BasePkgManager.install(
  File "c:\users\joshu\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 702:
    pkg_dir = self._install_from_url(name, url, requirements, track=True)
  File "c:\users\joshu\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 485:
    vcs = VCSClientFactory.newClient(tmp_dir, url)
  File "c:\users\joshu\.platformio\penv\lib\site-packages\platformio\vcsclient.py", line 45:
    obj = getattr(modules[__name__], "%sClient" % type_.title())(
  File "c:\users\joshu\.platformio\penv\lib\site-packages\platformio\vcsclient.py", line 61:
    self.check_client()
  File "c:\users\joshu\.platformio\penv\lib\site-packages\platformio\vcsclient.py", line 126:
    raise UserSideException(
============================================================================== [FAILED] Took 1.08 seconds ==============================================================================

Environment               Status    Duration
------------------------  --------  ------------
mega2560                  IGNORED
mega1280                  IGNORED
rambo                     IGNORED
FYSETC_F6_13              IGNORED
FYSETC_F6_14              IGNORED
sanguino644p              IGNORED
sanguino1284p             IGNORED
melzi                     IGNORED
melzi_optiboot            IGNORED
at90usb1286_cdc           IGNORED
at90usb1286_dfu           IGNORED
DUE                       IGNORED
DUE_USB                   IGNORED
DUE_debug                 IGNORED
LPC1768                   IGNORED
LPC1769                   IGNORED
STM32F103RC               IGNORED
STM32F103RC_fysetc        IGNORED
STM32F103RC_btt           IGNORED
STM32F103RC_btt_USB       IGNORED
STM32F103RC_btt_512K      FAILED    00:00:01.079
STM32F103RC_btt_512K_USB  IGNORED
STM32F103RE               IGNORED
STM32F103RE_btt           IGNORED
STM32F103RE_btt_USB       IGNORED
STM32F4                   IGNORED
STM32F7                   IGNORED
ARMED                     IGNORED
STM32F103VE_GTM32         IGNORED
STM32F103VE_longer        IGNORED
mks_robin_mini            IGNORED
mks_robin_nano            IGNORED
mks_robin                 IGNORED
mks_robin_pro             IGNORED
mks_robin_lite            IGNORED
mks_robin_lite3           IGNORED
jgaurora_a5s_a1           IGNORED
STM32F103CB_malyan        IGNORED
chitu_f103                IGNORED
STM32F401VE_STEVAL        IGNORED
FLYF407ZG                 IGNORED
FYSETC_S6                 IGNORED
STM32F407VE_black         IGNORED
BIGTREE_SKR_PRO           IGNORED
BIGTREE_GTR_V1_0          IGNORED
BIGTREE_BTT002            IGNORED
teensy31                  IGNORED
teensy35                  IGNORED
esp32                     IGNORED
linux_native              IGNORED
SAMD51_grandcentral_m4    IGNORED
rumba32_f446ve            IGNORED
rumba32_mks               IGNORED
include_tree              IGNORED
========================================================================= 1 failed, 0 succeeded in 00:00:01.079 =========================================================================
The terminal process "C:\Users\joshu\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

Typeopymp, 2020-08-15:
I get this same exact error.  Not sure what is causing it

EniSy, 2020-08-16:
Same thing pls. help/

oldfarmhand, 2020-08-20:
I get the same error message, I pulled the firmware file straight off of  github opened it put folder in c/btt opened the folder in vs code compiled it (WITHOUT ANY CHANGES)
and it failed.
I hope they answer you so I will know what to do.

thisiskeithb, 2020-08-21:
> UserSideException: Please install Git client from https://git-scm.com/downloads

Install git from the link in the error log & make sure you have Python installed.

CMendy, 2020-08-23:
https://github.com/CMendy/MarlinForBTT_SKR-Mini_E3_V2
I just created this repo with up to date Marlin firmware with SKR mini configuration.

Its the firmware I am using on my SKR-mini E3 V2.0

MTUpower, 2020-08-24:
Afternoon World, I am hoping that the community will be able to help with building and uploading firmware to the SKR MINI E3 V2.0 board by Bigtreetech.  This board Has a STM32F103RCT6 processor.  No matter what combination of marlin downloads and configuration files I try for a CR-10S or the Ender 3 I simply cannot get it to compile.  The error STM32F103RC_btt_512K FAILED always appears. I am at my Whitts end trying to figure this out and would appreciate help to guide me though this, before I take a hammer to the board!!

werto165, 2020-08-25:
> Afternoon World, I am hoping that the community will be able to help with building and uploading firmware to the SKR MINI E3 V2.0 board by Bigtreetech. This board Has a STM32F103RCT6 processor. No [...]

Hello, I've just purchased this myself got it working after I found this: 

https://reprap.org/forum/read.php?415,874307

Find and replace the line "platform = ststm32" with "platform = ststm32@<6.2.0" for the: 
```
[env:STM32F103RC_btt_512K]
platform          = ststm32@<6.2.0
board             = genericSTM32F103RC
board_upload.maximum_size=524288
platform_packages = tool-stm32duino
build_flags       = !python Marlin/src/HAL/STM32F1/build_flags.py
  ${common.build_flags} -DDEBUG_LEVEL=0 -std=gnu++14 -DHAVE_SW_SERIAL -DSS_TIMER=4 -DSTM32_FLASH_SIZE=512
build_unflags     = -std=gnu++11
extra_scripts     = buildroot/share/PlatformIO/scripts/STM32F103RC_SKR_MINI.py
src_filter        = ${common.default_src_filter} +<src/HAL/STM32F1>
lib_deps          = ${common.lib_deps}
  SoftwareSerialM=https://github.com/FYSETC/SoftwareSerialM/archive/master.zip
  USBComposite_stm32f1@==0.91
lib_ignore        = SPI
monitor_speed     = 115200

```

MTUpower, 2020-08-25:
Hi, many thanks for there reply, where exactly I do this change and do I copy and paste all that script?

werto165, 2020-08-25:
> Hi, many thanks for there reply, where exactly I do this change and do I copy and paste all that script?
> […](#)

sorry you just change the line in platformio.ini it'll be from: 
platform          = ststm32
to: 

platform          = ststm32@<6.2.0

MTUpower, 2020-08-25:
> … <x-msg://5/#>

Many thanks I will give it a try latter today.  Supposed to be working!!!!!

MTUpower, 2020-08-25:
> … <x-msg://6/#>

Sorry me again

Could not resist looking!!

Do you mean change it here? In the environment?

[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = STM32F103RC_btt_512K
include_dir  = Marlin

rosskinard, 2020-08-25:
Here's the two changes I made to my platformio.ini file and now both my SKR mini E3 1.2 and my 2.0 board both compile just fine.

`[platformio]
src_dir = Marlin
boards_dir = buildroot/share/PlatformIO/boards
default_envs = STM32F103RC_btt_512K
include_dir = Marlin`

and

`[env:STM32F103RC_btt_512K]
platform          = ${common_stm32f1.platform}
extends           = env:STM32F103RC_btt
board_upload.maximum_size=524288
build_flags       = ${env:STM32F103RC_btt.build_flags} -DSTM32_FLASH_SIZE=512
board_build.core  = maple`

 With these changes BTT firmware from this repo built fine.

I also converted the clean version of Marlin 2.0.6 to my BTT and got it to compile for both boards with the new stuff released. Takes a little while to convert your configuration.h and configuration_adv.h files over to the slightly changed format for the newer Marlin, but it works.

But with no changes other than the ones I put in above, both boards build from this repo.

alvarini, 2020-08-26:
hello, I followed the proposed solutions here. In my case I have Marlin 2.0.6 and SKR 1.2. The firmware is building with this warnings:

/Users/alvaromillan/.platformio/packages/framework-arduinoststm32-maple/STM32F1/libraries/STM32ADC/src/utility/util_adc.c:10:30: warning: 'adc_result' initialized and declared 'extern'
Compiling .pio/build/STM32F103RC_btt_512K/lib496/EEPROM/flash_stm32.c.o
 extern volatile unsigned int adc_result = 0;

/Users/alvaromillan/.platformio/packages/framework-arduinoststm32-maple/STM32F1/libraries/FreeRTOS701/utility/port.c: In function 'xPortStartScheduler':
/Users/alvaromillan/.platformio/packages/framework-arduinoststm32-maple/STM32F1/libraries/FreeRTOS701/utility/port.c:178:2: warning: implicit declaration of function 'systick_attach_callback' [-Wimplicit-function-declaration]
  systick_attach_callback(&xPortSysTickHandler);
  ^~~~~~~~~~~~~~~~~~~~~~~
At top level:
/Users/alvaromillan/.platformio/packages/framework-arduinoststm32-maple/STM32F1/libraries/FreeRTOS701/utility/port.c:285:6: warning: 'prvSetupTimerInterrupt' defined but not used [-Wunused-function]
 void prvSetupTimerInterrupt( void )

The firmware is flashed correctly and now working everything. My .ini is attached 
[platformio.ini.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5129626/platformio.ini.zip)

MTUpower, 2020-08-27:
Hi Again

Sorry to trouble but I am still having no luck with this.

Please find the errors below.  Have no idea what it all means.  I made the changes you suggested but still will not compile.

Marlin/src/module/motion.cpp:73:71: error: 'X_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                       ^
Marlin/src/module/motion.cpp:79:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
Marlin/src/module/motion.cpp:73:71: note: suggested alternative: 'HOMING_BUMP_MM'
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                       ^
Marlin/src/module/motion.cpp:79:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
Marlin/src/module/motion.cpp:73:80: error: 'Y_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                ^
Marlin/src/module/motion.cpp:79:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
Marlin/src/module/motion.cpp:73:80: note: suggested alternative: 'HOMING_BUMP_MM'
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                ^
Marlin/src/module/motion.cpp:79:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
Marlin/src/module/motion.cpp:73:89: error: 'Z_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                         ^
Marlin/src/module/motion.cpp:79:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
Marlin/src/module/motion.cpp:73:89: note: suggested alternative: 'HOMING_BUMP_MM'
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                         ^
Marlin/src/module/motion.cpp:79:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
Marlin/src/module/motion.cpp: In function 'void homeaxis(AxisEnum)':
Marlin/src/module/motion.cpp:1567:27: error: 'Z_HOME_BUMP_MM' was not declared in this scope
       (axis == Z_AXIS && (Z_HOME_BUMP_MM)) ? _MAX(Z_CLEARANCE_BETWEEN_PROBES, Z_HOME_BUMP_MM) :
                           ^~~~~~~~~~~~~~
Marlin/src/module/motion.cpp:1567:27: note: suggested alternative: 'HOMING_BUMP_MM'
       (axis == Z_AXIS && (Z_HOME_BUMP_MM)) ? _MAX(Z_CLEARANCE_BETWEEN_PROBES, Z_HOME_BUMP_MM) :
                           ^~~~~~~~~~~~~~
                           HOMING_BUMP_MM
*** [.pio/build/STM32F103RC_btt_512K/src/src/module/motion.cpp.o] Error 1
================================================================ [FAILED] Took 69.29 seconds ================================================================

Environment               Status    Duration
------------------------  --------  ------------
mega2560                  IGNORED
mega1280                  IGNORED
rambo                     IGNORED
FYSETC_F6_13              IGNORED
FYSETC_F6_14              IGNORED
sanguino644p              IGNORED
sanguino1284p             IGNORED
melzi                     IGNORED
melzi_optiboot            IGNORED
at90usb1286_cdc           IGNORED
at90usb1286_dfu           IGNORED
DUE                       IGNORED
DUE_USB                   IGNORED
DUE_debug                 IGNORED
LPC1768                   IGNORED
LPC1769                   IGNORED
STM32F103RC               IGNORED
STM32F103RC_fysetc        IGNORED
STM32F103RC_btt           IGNORED
STM32F103RC_btt_USB       IGNORED
STM32F103RC_btt_512K      FAILED    00:01:09.286
STM32F103RC_btt_512K_USB  IGNORED
STM32F103RE               IGNORED
STM32F103RE_btt           IGNORED
STM32F103RE_btt_USB       IGNORED
STM32F4                   IGNORED
STM32F7                   IGNORED
ARMED                     IGNORED
STM32F103VE_GTM32         IGNORED
STM32F103VE_longer        IGNORED
mks_robin_mini            IGNORED
mks_robin_nano            IGNORED
mks_robin                 IGNORED
mks_robin_pro             IGNORED
mks_robin_lite            IGNORED
mks_robin_lite3           IGNORED
jgaurora_a5s_a1           IGNORED
STM32F103CB_malyan        IGNORED
chitu_f103                IGNORED
STM32F401VE_STEVAL        IGNORED
FLYF407ZG                 IGNORED
FYSETC_S6                 IGNORED
STM32F407VE_black         IGNORED
BIGTREE_SKR_PRO           IGNORED
BIGTREE_GTR_V1_0          IGNORED
BIGTREE_BTT002            IGNORED
teensy31                  IGNORED
teensy35                  IGNORED
esp32                     IGNORED
linux_native              IGNORED
SAMD51_grandcentral_m4    IGNORED
rumba32_f446ve            IGNORED
rumba32_mks               IGNORED
include_tree              IGNORED
=========================================================== 1 failed, 0 succeeded in 00:01:09.286 ===========================================================
duncanstickland@Duncans-iMac Marlin-2.0.x-SKR-mini-E3-V2.0 %

ehansen2580, 2020-09-02:
> Hi Again Sorry to trouble but I am still having no luck with this. Please find the errors below. Have no idea what it all means. I made the changes you suggested but still will not compile. [...]

Did you have success in finding a solution here?  This is the error I'm coming up with now after moving the files to a shorter path name and making sure the git was installed.

MTUpower, 2020-09-02:
> Hi Again Sorry to trouble but I am still having no luck with this. Please find the errors below. Have no idea what it all means. I made the changes you suggested but still will not compile. Marlin/src/module/motion.cpp:73:71: error: 'X_HOME_BUMP_MM' was not declared in this scope #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval NAME##P = { X##OPT, Y_##OPT, Z_##OPT } ^ Marlin/src/module/motion.cpp:79:1: note: in expansion of macro 'XYZ_CONSTS' XYZ_CONSTS(float, home_bump_mm, HOME_BUMP_MM); ^~~~~~~~~~ Marlin/src/module/motion.cpp:73:71: note: suggested alternative: 'HOMING_BUMP_MM' #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval NAME##P = { X##OPT, Y_##OPT, Z_##OPT } ^ Marlin/src/module/motion.cpp:79:1: note: in expansion of macro 'XYZ_CONSTS' XYZ_CONSTS(float, home_bump_mm, HOME_BUMP_MM); ^~~~~~~~~~ Marlin/src/module/motion.cpp:73:80: error: 'Y_HOME_BUMP_MM' was not declared in this scope #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval NAME##P = { X##OPT, Y_##OPT, Z_##OPT } ^ Marlin/src/module/motion.cpp:79:1: note: in expansion of macro 'XYZ_CONSTS' XYZ_CONSTS(float, home_bump_mm, HOME_BUMP_MM); ^~~~~~~~~~ Marlin/src/module/motion.cpp:73:80: note: suggested alternative: 'HOMING_BUMP_MM' #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval NAME##P = { X##OPT, Y_##OPT, Z_##OPT } ^ Marlin/src/module/motion.cpp:79:1: note: in expansion of macro 'XYZ_CONSTS' XYZ_CONSTS(float, home_bump_mm, HOME_BUMP_MM); ^~~~~~~~~~ Marlin/src/module/motion.cpp:73:89: error: 'Z_HOME_BUMP_MM' was not declared in this scope #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval NAME##P = { X##OPT, Y_##OPT, Z_##OPT } ^ Marlin/src/module/motion.cpp:79:1: note: in expansion of macro [... cut ...]  rumba32_f446ve IGNORED rumba32_mks IGNORED include_tree IGNORED =========================================================== 1 failed, 0 succeeded in 00:01:09.286 =========================================================== duncanstickland@Duncans-iMac Marlin-2.0.x-SKR-mini-E3-V2.0 %
> … <x-msg://17/#>

Hi Ehansen2580

I am sorry to say NO I have not found the solution.  I have been at it for over a week, trying everything under the sun, reading and watching any article or video I could get my hands on.  All the kind suggestions from many people have been to no avail.  So I have given up, I took the Hammer to the board last night, and thrown the remains in the bin!!

ehansen2580, 2020-09-02:
Well, I hope that isn't really the case but it is very frustrating.

> Hi Ehansen2580 I am sorry to say NO I have not found the solution. I have been at it for over a week, trying everything under the sun, reading and watching any article or video I could get my hands [...]

MTUpower, 2020-09-02:
A am afraid to say its true.

casterle, 2020-09-03:
> Install git from the link in the error log & make sure you have Python installed.

I've updated git and Python 3.8.5 was already installed. When I build I'm getting errors such as:

`C:\Users\casterle\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\USBComposite\usb_x360.c:486:1: fatal error: opening dependency file .pio\build\STM32F103RC_btt_512K\lib07d\USBComposite\usb_x360.c.d: No such file or directory
 }
 ^
compilation terminated.`

even though the file "usb_x360.c.d" exists where specified in my build tree.

I had no problems building an earlier version of this code, and I just downloaded and built Marlin 2.0.6.1 from github earlier today.

Koyodee, 2020-09-04:
Unrelated to the problem, but I have a similar error, and was kinda freaked out since mine is located in C:\Users\joshu as well lol

ehansen2580, 2020-09-10:
I finally got mine compiling.
Had to install git
Then I deleted all copies of the original download and got a fresh copy
I extracted it to a directory right below C:\
Got new copies of the configuration files

Used vscode and everything worked

MTUpower, 2020-09-10:
> … <x-msg://2/#>

Morning World!!

Last night dear old Amazonian delivered a new SKR Mini E3 V2 control board. You remember I took the hammer to my last one. Anyway with renewed vigour and enthusiasm I have at last uploaded run and tested it on my CR-10.  OHMG its so quiet!  To do this I used the latest rendition of the Marlin firmware, and configuration files.  I used the Ender 3 files and then pasted them, as you do, then changed the settings for a CR-10 within the firmware. After all a CR-10 is basically a large Ender 3.  This I have tested the set up, prints extremely well, did I say its quiet!!!  

I still have a small issue to solve to which I would be great full for some help!  When I do the level all corners it thinks the bed is the ender 3 size?  But when probing the bed with the BLtouch it knows it has the 300 x 300 bed, as these are the parameters that I entered for bed size. Why?

Cheers all.