bigtreetech/BIGTREETECH-SKR-mini-E3 #365: [BUG] (short description)

electricianbible, 2020-08-24:
### Description

I am trying to install this board (BIGTREETECH SKR Mini E3 V2.0) on my ender 3 v2.

This is my configuration files.
when try to pre heat pla and bed get this error

Heating Failed: E1
Printer halted
Please reset

Also try the BIGTREETECH firmware files and won’t work same error. Any help would be really appreciated.

My Configurations
Configuration.h
https://drive.google.com/file/d/1mzNn8rFFu126a-psCAQTmuaL2_EncWAe/view?usp=sharing
Configuration_adv.h
https://drive.google.com/file/d/13GXvZy4t8r_c8DJOfwwtiE0glVNOopKx/view?usp=sharing

Steps to Reproduce
install BIGTREETECH firmware
try to preheat main > temperature >preheat pla
not change in temperature not voltage is going to the bed or the hotend.

NOTE:THIS IS IN A ENDER-3 V2

brew99, 2020-08-28:
Check all your heating and thermistor connections to the board.   Make sure they are in the correct place.  Do you have a voltmeter to check if voltage is getting to the E1 connection when activated?

electricianbible, 2020-08-29:
well if i put back the original ender e v2 board everything is working fine.
there is not voltage coming out off neater hotend or bed
if i heat any of the thermistors the display show variation on the readings. what makes me believe that there are doing their function.
the only thing that i think I have the problem is on the buck converter for the fans. I got 3 of them.