bigtreetech/BIGTREETECH-SKR-mini-E3 #366: [Missing Include path] (c_cpp_properties.json)

quarinteen, 2020-08-25:
### Description

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

alvarini, 2020-08-26:
same for me... I tried updating platformio.ini but no luck. If I build a firmware with 2.0.6 version of Marlin my printer gets freezed. 
*** [.pio/build/STM32F103RC_btt_512K/FrameworkArduino/avr] /Users/xxxx/Documents/Marlin-2.0.x-SKR-Mini-E3-V1.2/.pio/build/STM32F103RC_btt_512K/FrameworkArduino/avr: No such file or directory
/Users/xxxx/.platformio/packages/framework-arduinoststm32-maple/STM32F1/variants/generic_stm32f103r/wirish/start_c.c:95:1: fatal error: opening dependency file .pio/build/STM32F103RC_btt_512K/FrameworkArduinoVariant/wirish/start_c.c.d: No such file or directory

brew99, 2020-09-02:
Put your Marlin files closer to the root of the drive and try compiling.   There is a limitation of 256 characters (I think?) in a path