bigtreetech/BIGTREETECH-SKR-mini-E3 #367: Marlin 2.0.6 Skr Mini E3 V2.0 SVC Code Probleme for Ender 3

Orcanis86, 2020-08-26:
[image]
How can i get fix this probleme?

Orcanis86, 2020-08-26:
[image]
[image]

Orcanis86, 2020-08-26:
[1 failed.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5130087/1.failed.txt)

Orcanis86, 2020-08-26:
[Configuration.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5130505/Configuration.txt)

[Configuration_adv.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5130507/Configuration_adv.txt)

Orcanis86, 2020-08-27:
[image]

brew99, 2020-08-28:
Try compiling the Marlin Bugfix-2.0.x version

Orcanis86, 2020-08-28:
> 
> 
> Try compiling the Marlin Bugfix-2.0.x version

I've done that too now. But it still does not work. With 2.0.x I get the error messages too...

[image]

CrysKilljoy, 2020-09-08:
Same for me, can't compile it without even changing a thing