bigtreetech/BIGTREETECH-SKR-mini-E3 #369: [BUG] change filament

redgod, 2020-08-28:
Hi, I'm learning how to compile, I have an ender 3 pro with skr mini 1.2 + tft24 v1.1, my problem recurs when I change the filament, when the print returns, it does 2 layers above, I tried several tips and none of them could check my code. I am with marlin 2.0.6 to observe the problem? Thank you

neither does it work with the official btt code

ps: sorry for my english 

### Steps to reproduce

1. [compiles correctly]
2. [installs correctly]
3. [filament change is activated]

**Expected behavior**
that when returning to print after the change of filament do it to the correct layer

**Actual behavior**
when he returns to continue printing he does it 2 layers above 

[image]

[Marlin-2.0.6.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5142162/Marlin-2.0.6.zip)

brew99, 2020-08-28:
Try compiling the marlin Bugfix-2.0.x and see if it is the same issue.

redgod, 2020-08-28:
The same happens, if someone checks if my code is correct I would appreciate it

ysoline, 2020-12-13:
I have similar problem :(

Aeryzor, 2021-01-29:
I'm having the same issue. Only on CURA tho, on Prusa it continues printing without any problem but I really don't like Prusa haha