bigtreetech/BIGTREETECH-SKR-mini-E3 #371: Neopixel wont compile

pRINTERnOODLE, 2020-08-28:
### Description

Cant get Neopixel ativated using latest bugfix 2.0.6
### Steps to reproduce

1. [First Step] Activate Neopixel support
// Support for Adafruit NeoPixel LED driver
#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRB // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
 // #define NEOPIXEL_PIN     PA8       // LED driving pin
  //#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
  //#define NEOPIXEL2_PIN    5
  #define NEOPIXEL_PIXELS 8       // Number of LEDs in the strip. (Longest strip when NEOPIXEL2_SEPARATE is disabled.)
  #define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 127  // Initial brightness (0-255)
  #define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup
2. [Second Step] compile
3. [and so on...] fail

**Expected behavior**

Working Neopixels
**Actual behavior**

Failure to compile
### Additional Information
Have tried most suggestions in the threads regarding this problem including changing and adding several different references to NeoPixel repositories
NEOPIXEL_LED            = Adafruit NeoPixel@1.5.0
                          src_filter=+<src/feature/leds/neopixel.cpp>
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

freak12321, 2020-09-24:
I can confirm this, same steps to reproduce.
It is not possible to compile when Neopixel enabled.

Piscanc, 2020-10-04:
I needed to install git for windows, then it would compile.

joshwapohlmann, 2020-10-31:
Had the same issue at first.
BUT I was using a clone from the **official marlin repository**.
Changes/patches I applied to make it compile where:
1. Copied the complete `platformio.ini` from this repo here. Especially in line 719 the modified library for this platform is defined. BTT did some [changes](https://github.com/bigtreetech/Adafruit_NeoPixel/commit/b72627a0042055ebe223e605c3d3cfd524c7ab47) to the original driver to support STM32F1.
2. Disabled `#define NEOPIXEL_PIN` in `Configuration.h` as it is redefined by board specific code/header
3. Disable the sanity check in `/Marlin/src/HAL/STM32F1/inc/SanityCheck.h#53` as it no longer applies with the custom version of the driver.

No it should compile.
OR else: Just use THIS repository :man_shrugging:

877dev, 2020-11-20:
Having this exact same issue with latest Marlin bugfix-2.0.x and config, using SKR Mini E3 v2.0 board.

Any fixes?

877dev, 2020-11-20:
> 2. Disabled `#define NEOPIXEL_PIN` in `Configuration.h` as it is redefined by board specific code/header

@joshwapohlmann thanks, could you expand on this please?
I find that just copying the `platformio.ini` from here make it compile with latest Marlin bugfix-2.0.x, without disabling the sanity check. 

If  `#define NEOPIXEL_PIN`  is to be commented out, how do you define pin (PA8 ?) and number of LED's?

joshwapohlmann, 2020-11-24:
@877dev There is a board specific configuration loaded for the SKR mini E3 which redefines NEOPIXEL_PIN. Therefore it should through an error or warning that it is redefined.

877dev, 2020-11-24:
> @877dev There is a board specific configuration loaded for the SKR mini E3 which redefines NEOPIXEL_PIN. Therefore it should through an error or warning that it is redefined.

Thank you,I did notice that during compile actually. I'll give it a try 👍