bigtreetech/BIGTREETECH-SKR-mini-E3 #374: [BUG] PWM Fans are noisy

SirRonBird, 2020-09-01:
### Description

Both of the PWM controlled Fans are quite noisy, by that i mean there is this typical pwm noise. 
I use the original Ender 3 electronic-case-fan and as the part-cooling-fan a DC 12V/24V 2 Pin fan. Both are directly connected on the board.
I use the precompiled firmware-bltouch.bin.

**Expected behavior**
Fans should be much quiter

**Actual behavior**
they aren´t quite

DanielYWoo, 2020-12-29:
the board fan is very noise when pwm is from 4% to 99%。same here

thegwydd, 2021-03-23:
I have the same issue.
@DanielYWoo, can you explain better what you mean by "swap the fan"? In my case the part cooling fan is connected to the proper connector as I can drive it using marlin menu.
Are you talking about Vcc and Gnd?

DanielYWoo, 2021-03-23:
@thegwydd  No. I am NOT saying VCC/GND swap. You can check the SKR mini E3 schematic, the layout of the two fan connectors are different from the stock board.