bigtreetech/BIGTREETECH-SKR-mini-E3 #375: [BUG] doesn’t fit : STM32F103RC_btt_512K !=  STM32F103RC size

erichards-nvidia, 2020-09-01:
#102 Description

STM32F103RC_btt_512K is used which indicates that there is 512kb if flash but the board uses a STM32F103RC which is 256kb. 

If you build with the correct board STM32F103RC_btt[_USB] definition with a lot of features turned on like graphical UI, WiFi, usb memory stick, bltouch, G2/G3 ARC support it doesn’t fit. 
If you build with 512kb board type it’ll build but not flash coz it’s too big. 

So how do you get all the advanced features the SKR mini E3 brings?
I want to upgrade to get G2/G3, BLtouch, Filament runout, WiFi, usb host and graphics panel but if it won’t build or flash what’s the point?

**Expected behavior**
Build with 512kb flash and have a 512kb cpu on the product

**Actual behavior**
Doesn’t fit on the Mini E3 v2 cpu.

erichards-nvidia, 2020-09-02:
> You are correct in theory that the 512k shouldn't fit on the RC chip, but there have been multiple reports where the RC label may have been put on a chip that came from the the RE die. exactly why this was done, who knows, but it does work (I'm running the 512K on the Rc chip, but I did need to update the bootloader on my V1.0 board).
> 
> Have you tried compiling with the 512K?

Yup it compiles great with 512k - with everything turned on I want (ARC_SUPPORT, BLtouch, Softstops, USB, SDcard, WiFi, Oled Gfx)... and it comes out to about 30% RAM and 65% FLASH but I'm concerned it's not going to burn on a board if I buy one coz it's an RC not an RE.
Do we know for a fact that the Mini E3 v3 Bill Of Material (BOM) is an RE? Coz the Schematic says RC (ie 256k Flash).

Are you sure your build image didn't happen to end up under 256K? I ask coz I was able to make a stock board features clone in less than 256k. But I couldn't get ARC_SUPPORT (G2/G3) and some other non-add-in-card-features to build small enough.

brew99, 2020-09-02:
There is no Mini E3 v3...newest is a V2.   That being said, I don't know if it actually has a RE or RC, but the sch. says RC.   The V1.0 and V1.2 were both RC chips.

If I built the image under 256K (minus the EEPROM size, etc), it would go on my board without updating the bootoloader.   With the updated bootloader (from a V1.2) if worked with under or over 256k, no issues.

I beleive the V1.2 had no issues with over 256K, as the bootloader was not locked to that size.

V2.0 should be fine, as many have compiled with that 512K setting

erichards-nvidia, 2020-09-02:
sorry typo - I meant E3 v2.
Ok I have enough confidence now and will simply get an E3 v2 board with the assumption that it's actually BOMed with an RE chip despite the schematic being RC. If it's pin compatible I can see why they didn't bother to update the symbol. It just left me wondering what's up. It would have been less confusing if they had called the board package name STM32F103R**E**_btt_512K - then there would have been no mismatch.
thx so much for replying.

erichards-nvidia, 2020-09-03:
I will file a Feature Request to fix this confusion

closed 2020-09-03 by erichards-nvidia