bigtreetech/BIGTREETECH-SKR-mini-E3 #377: This board sucks, No compile for SKR mini E3 V2.0[BUG] (short description)

dodgedoc, 2020-09-02:
BTT  You release a board that does not even compile under VSCode.  What?????  I need help with this or you can come get your board and screens I have got.  I can download code for TH3D Easyboard and it complies fine.. What is up with you guys.. If it in platformIO then update it.. This is letting ppl down.. 
This is just my thoughts... Barry

casterle, 2020-09-03:
I just noticed I can no longer build the code I downloaded today with VSCode either. The previous version did build so you might try that.

tablatronix, 2020-09-09:
Whats the error? I compile under pio fine.

pbsuper, 2020-09-10:
if you just get a fresh firmware file from marlin and set it up for this board so the right CPU and such it does compile but the BTT Tree just sucks A.T.M.

tablatronix, 2020-09-10:
oh the repo has problems?

pbsuper, 2020-09-10:
Yes they are using the wrong lib for neopixel with this cpu so. You can't compile it cause of that

tablatronix, 2020-09-10:
Good to know , I am about to add my neopixels tonight

dodgedoc, 2020-09-10:
I got a file from CMendy, It would not compile either. The under platformIO, i found under quick access that you can update all libraries so i updated them. Not many out of date. Tried to compile after that, no go. Back to quick access, found upgrade platformIO core. I upgrade the core? dont know what it is. Then i tried to compile after upgrading. Guess what it worked.
But will not compile with the file from BTT. 
Hope that helps someone out.. 
Barry

tablatronix, 2020-09-10:
I did just see a neopixel issue earlier when I was browsing closed issues

btolarz, 2020-09-12:
As workaround copy https://github.com/MarlinFirmware/Marlin/tree/2.0.x/buildroot and then compile. Last commit on this repo broke everything.