bigtreetech/BIGTREETECH-SKR-mini-E3 #379: [Part Fan Not Updating Immediately] When setting fan speed using G106, the fan doesn't update immediately.

therooftopprinz, 2020-09-03:
When setting fan speed using M106, the fan doesn't update immediately even though speed has already updated in info screen.
The fan speed will be updated only when screen is changed using input interface.

### Steps to reproduce

1. Initially fan is off, set part fan speed to 100% using gcode: M106 P0 S255.
2. Fan speed has been updated in the info screen. But the fan is still off. Navigate to menu (push the knob), fan will start spinning at the set speed.

This also similar when controlling the speed via the interface:
1. Push knob to access menu, navigate to Temperature > Fan Speed.
2. Set fan speed. (Doesn't update fan speed in real time).
3. Navigate back (Temperature page), the fan speed isn't still updated.
3. Navigate back (Main menu page), the fan speed is now set.  

### Additional Information
https://pastebin.pl/view/8d70b012 Configuration.h
https://pastebin.pl/view/24441cc4 Configuration_adv

dado991, 2020-11-03:
Same problem here with SKR Mini E3 v2, and BTT firmware (no bltouch) at v2.0.7.2

Any suggestion on how to fix this?

dado991, 2020-11-03:
I think this is the same issue: [Issue 430](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/430)

Cathprotech, 2020-11-10:
I updated my firmware last night (9th Nov 2020) and only came to print this evening. Got exact same problem, part cooling fan not running but says 100% on display. I searched on the internet and found several threads on here with same problem. I clicked the button as someone mentioned and away it went. 
So, will anyone at BTT re-compile the firmware with the fix please. Compiling my own is a bit beyoned me, not the compiling part but the mass of options that I just don't understand.
One good thing about this update, is that I can now do Mesh Bed Leveling which I could not do before on the last version, so don't want to lose that option by rolling back.

onebehler, 2020-12-14:
#459 is what I feel I am seeing with this issue