bigtreetech/BIGTREETECH-SKR-mini-E3 #381: Firmware firmware-bltouch-for-z-homing.bin for E3 v1.2 has no leveling

ophbalance, 2020-09-03:
The firmware from 9/1 is about 10k smaller than the previous compile and doesn't appear to have any BL Touch options in the menus.

warlord2000, 2020-09-08:
yes its true

DTL59, 2020-09-09:
I encountered the same issue.

dokcal, 2020-09-10:
Yes, I just noticed the same thing. Darn near wrecked my printer!

bond3ll, 2020-09-11:
Same issue, please fix this!

abell529, 2020-09-15:
How do you get the previous firmware? I'd like to have the BLTouch options now that I have one installed.... there is no BLTouch menu option.

ophbalance, 2020-09-15:
> How do you get the previous firmware? I'd like to have the BLTouch options now that I have one installed.... there is no BLTouch menu option.

If you look at the history on the .bin file you can see all the previous commits.  I want back to the Apr 19 commit if memory serves.

abell529, 2020-09-15:
> > How do you get the previous firmware? I'd like to have the BLTouch options now that I have one installed.... there is no BLTouch menu option.
> 
> If you look at the history on the .bin file you can see all the previous commits. I want back to the Apr 19 commit if memory serves.

Thank you! I can't believe I missed that...

Shailevy, 2020-09-17:
Same issue... Horrible version but they don't seem to be paying attention this version

itsmetheuglykid, 2020-09-23:
Yup, damn near wrecked my printer.

PrutzerNL, 2020-09-23:
I tried all the V1.2 firmware-bltouch-for-z-homing firmwares. They all have problems. I'm now using the  SKR mini E3 V1.2 - Ender 3 BLtouch + filament runout firmware from teachingtech (create youtube channel): https://gumroad.com/teachingtech?recommended_by=library#HFxSQ

Finally everything works as it shoot be.
The SKR mini boards are create. But the official firmware is not good and needs attention!