bigtreetech/BIGTREETECH-SKR-mini-E3 #383: Can anyone help me with firmware?

spoonsarefun, 2020-09-06:
I don't have the Facebook and would really appreciate it if someone could help me out as I've spent a lot of time and don't seem to get far just from one problem to next.  I purchased this board for my ender 5 pro that I just purchased as it's 32bit and after reading advantages I was sold but now that I have it and the new touchscreen that plugs in I can't get anything to seemingly worth as intended.  
It is a ender 5 pro with the sir mini e3 v1.2 installed.  I also have the tft 3.5 e3 v30.  I also have a microswiss direct drive if that is relevant and bt touch.  The bt touch is hanging but not hooked up I lost most the day trying to get the motherboard and screen working. 

 The touchscreen won't detect printer and I've tried baud rate to no avail but detects in emulated.

I managed to compile a firmware following instructions and the bed sets it self and preheats than runs quickly from top right to bottom left and tries to make a run off the printer and self destruct.  

I have not hooked up with autobed level yet and have little time until next weekend but would appreciate if someone could help me out with compiling a firmware with and without bed level that would allow me to print again 😁.

jwg4096, 2020-09-27:
For help like this, I would suggest downloading the Reddit app to your handset, or using the Reddit site from your PC.  There are multiple subreddits that lots of people contribute to, answering questions exactly like yours, you just subscribe to them:

- r/ender5
- r/bigtreetech
- r/3dprinting

I have asked loads of questions on there and had quick responses.  Very good 3D printing community.

Good luck!

spoonsarefun, 2020-11-11:
Haha very good.  I will go searching again thanks for telling me to Google like I had never done it before.  The problem with these googled instructions are half are half-baked others don't include specific things specific to your printer and unless you are knowledgeable in printing already or want to spend hours on hours on it which I already had hence asking for help but thanks for the assist and wish you the best I guess I have a paperweight or a poorly printing machine until I get it resolved.  Figured being it's bigtreetech and supposed to be a direct drop in upgrade for ender 5 pro it would be ..but it has turned into anything but.  

Can you link the forum with the exact answer? Vs generic search results?
  Is this is bigtreetech support site or the redirect you to Reddit for support site?