bigtreetech/BIGTREETECH-SKR-mini-E3 #384: Skr mini e3 v2.0 yaxis problem

xXWSDCXx, 2020-09-07:
Hello,
I have a problem with the skr mine e3 v2.0 and the yaxis of my ender 3.

The bed is trying to drive to far to the front while the ABL process is running. It hits the belttensioner everytime. I tried to fix it with homeoffset and probe offset, but no chance to fix it.
I also tried to compile a new marlin, but then I can't activate ABL with a bltouch for the probepins. Idk why.

Please help. 

Sincerely 
xXWSDCXx

xXWSDCXx, 2020-09-07:
[HwVideoEditor_2020_09_07_165202.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5184361/HwVideoEditor_2020_09_07_165202.zip)

jtroberts1, 2020-09-09:
I am having the exact same issue.  Additional info...  Using BLtouch.  and my pin configuration is example 2 where I am not using the 5 pin header.  I am only using  3 pins of the BLtouch + 2 pins in the Z stop.  the steps described here...   https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/bltouch-2.0-2.jpg

ihagens, 2020-09-21:
I'm having the same issue. its working fine with the older firmware. (which has other bugs)

Thegs68, 2021-03-13:
I am having the same issue. Anyone found a solution yet?_