bigtreetech/BIGTREETECH-SKR-mini-E3 #385: [BUG] (short description)

spirit527, 2020-09-07:
### Description

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

spirit527, 2020-09-07:
Like many others on this post I had a working version of the skr mini E3 v2 and was very happy with this but decided to add a generic bltouch and flash the board with bltouch firmware. My screen turned blue and even though I tried to reinstall skr mini firmware it hasn't changed.
Can someone attach the original masterfile that came with this board because I never kept the file, and the files I download on the github don't seem to fix this issue.

Lotek213, 2020-09-07:
Please find attached original firmware (without support of bltouch) and also firmware with X-homning using bl-touch.
If you have TFT35 display or other serial device connected to the board, please disconnect it before flashing, because serial port could interfere with firmware flashing process.
I had the same issue and I though that I brick my board. I tried again without serial connection to TFT and board started working again.   

[firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5184888/firmware.zip)