bigtreetech/BIGTREETECH-SKR-mini-E3 #388: BLTouch option not showing in printer menu

DecXdrones, 2020-09-11:
I tried downloading the firmware-bltouch-for-z-homing.bin and flashing it into my ender 3 but everytime I try to do so I can't find the bltouch fonction in my printer's menu. I've checked my connections and everything looks fine. I don't know if I did something wrong or if the problem is actualy coming from the firmware.

I'd like to know if there is a new version coming soon to fix this problem or if anyone could just explain me what is wrong with my method.

hongmoondo, 2020-09-14:
Exactly same for my case.

Near76, 2020-09-14:
Same issue here, I also cannot get it to work properly.  This is a first time install for me.  Do I connect the z stop wires (white/black) next to the BLT probe header pin or in the old location of the z stop switch?  The readme in the V2.0 shows pics of both ways, but is unclear which one I should be using with the z-homing.bin FW.

DecXdrones, 2020-09-15:
@Near76 it depends on which firmware you decide to upload in your printer. If you want to keep your stock Z switch and only use the BL Touch to do the bed mapping you need to keep your z switch where it is normally and add your black/white connector on the probe plug beside the servos switch. Once this is done you need to use the firmware-bltouch.bin file to upload in the printer. On the other hand if you want to use the BL Touch for the homing too you need to remove the Z switch connector and replace it with your black/white connector from the BL Touch this way the BL Touch will trigger and tell the printer to stop lowering. If you do so you will need to use the firmware-bltouch-for-z-homing.bin file instead. In both cases make sure to rename the file to firmware.bin before loading in the printer or the printer won't recognize the filename. The problem here is that once we uploaded the firmware-bltouch-for-z-homing.bin file, the BL Touch menu isn't showing in the menu anymore so it's impossible to go and finish the setting of the sensor.

abell529, 2020-09-15:
I'm having the same issue -- when the printer starts up, the BLTouch lights up and does the test clicking, but the option is not available in the menu.

PrutzerNL, 2020-09-22:
Same problem. I use SKR mini V1.2, firmware-bltouch-for-z-homing.bin.
Bltouch options are missing. Is it possible to fix this?

DecXdrones, 2020-09-22:
I just tried uploading the new firmware that came out on september 21 2020 and the same problem remains.

itsmetheuglykid, 2020-09-23:
I'm having the same issue. 6 hrs of fiddling and I'm getting frustrated. Anyone have any ideas?

itsmetheuglykid, 2020-09-23:
Nevermind, i figured it out. I used the firmware-bltouch-for-z-homing.bin but the current version does not have bed leveling. What you need to do is grab an older version. Click firmware.bin. Then click the History button with the clock symbol next to it in top right corner. Scroll down to the the April 19th commit. Then click 'browse the repository from this point in the history'...the button looks like this '<>'. Then grab the required firmware, change the name to firmware.bin and put it on your TF card, pop that bad boy in the printer and you should be good.

DecXdrones, 2020-10-05:
@itsmetheuglykid It works !!! Finally after 8 months I'll be able to setup my bltouch and print. Thank you Sir you juste made my day