bigtreetech/BIGTREETECH-SKR-mini-E3 #389: Latest firmware stepper motors much louder?

urshurak776, 2020-09-11:
Just downloaded the latest firmware for my SKR Mini E3 V2.0 board and the unit is much louder than the 2.0 drivers and feature set seems a lot smaller in the menus.  Thinking of going back.

davidv3rs, 2020-09-12:
Came here to say this. Also baby stepping doesn't seem to work. Get an "Unknown command: M290..." error on the display. Was going to open a new issue but I'll just go back to 2.0 bugfix.

chiibis, 2020-09-12:
Same issue here. Have to install previous version because of noise.

Halvehak, 2020-09-12:
Same issue here. Installed latest 2.0.6 firmware and now the stepper motors making a grinding noise.

davidv3rs, 2020-09-12:
@chiibis where do you get the previous version source?

davidv3rs, 2020-09-12:
Found the fix:

Configuration_adv.h
Uncomment lines 2407-2409

```
  /**
   * TMC2130, TMC2160, TMC2208, TMC2209, TMC5130 and TMC5160 only
   * Use Trinamic's ultra quiet stepping mode.
   * When disabled, Marlin will use spreadCycle stepping mode.
   */
  #define STEALTHCHOP_XY
  #define STEALTHCHOP_Z
  #define STEALTHCHOP_E
```

Looks like Stealth Chop was turned off.

urshurak776, 2020-09-12:
Thanks David!   Wonder why they changed it?

urshurak776, 2020-09-12:
What does M502 and M500 do?

davidv3rs, 2020-09-12:
> What does M502 and M500 do?

Resets your EEPROM.

urshurak776, 2020-09-12:
Thanks David!

urshurak776, 2020-09-13:
So, I downloaded the latest Marlin 2.0.6.1.  Followed the guide on reddit For the SKR Mini E3 2.0 and watched video from Chris Riley (Chris’ basement in youtube.). Compiled it, exported the bin file and flashed it to the board. Everything worked great EXCEPT my computer would recognize the board when plugged into my PC.  Drivers were fine.  Just to check, I flashed back to 2.0.X bug fix and my computer recognized it again.  Checked the serial port is -1 and serial port 2 is 2.  Not sure.  Good news is the stepper drivers were nice and quiet again, but I need functionality with the computer.  

I also tried overwriting the config.h and advanced.h files  with the ones from BTT’s site but it wouldn’t compile.  Not sure why.  So, for now, sticking with the 2.0.x bugfix.

davidv3rs, 2020-09-13:
> So, I downloaded the latest Marlin 2.0.6.1. Followed the guide on reddit For the SKR Mini E3 2.0 and watched video from Chris Riley (Chris’ basement in youtube.). Compiled it, exported the bin file and flashed it to the board. Everything worked great EXCEPT my computer would not recognize the board when plugged into my PC. Drivers were fine. Just to check, I flashed back to 2.0.X bug fix and my computer recognized it again. Checked the serial port is -1 and serial port 2 is 2. Not sure. Good news is the stepper drivers were nice and quiet again, but I need functionality with the computer.
> 
> I also tried overwriting the config.h and advanced.h files with the ones from BTT’s site but it wouldn’t compile. Not sure why. So, for now, sticking with the 2.0.x bugfix.

Hmm, I'll look into it.

urshurak776, 2020-09-13:
Think I figured it out.  In the platformio.ini file I didn’t Change default_envs = STM32F103RC_BTT_512K. 

I should have copied all three files over and made your changes on the advanced configuration file.  I’ll try it in morning and report back. 

Thanks David.

chiibis, 2020-09-13:
> @chiibis where do you get the previous version source?

Not sources, just previous `firmware.bin` file. 

From history on github:

[image]

davidv3rs, 2020-09-13:
> > @chiibis where do you get the previous version source?
> 
> @davidv3rs Not sources, just previous `firmware.bin` file.
> 
> From history on github:
> 
> [image]

Ahh, good to know. Thank you!

zurtox, 2020-09-17:
Yes, I'm facing the same issue with the Mini v1.2

Disk1of5, 2020-09-21:
Same issue.

timqui, 2020-09-22:
I see an update 18 hours ago for Ender 3 firmware, but I didn´t try it.

 [https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V2.0](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V2.0)

Disk1of5, 2020-09-22:
I can't test until the binaries for the firmware-bltouch-for-z-homing.bin drop.

urshurak776, 2020-09-23:
Looks like the Stealth Chop is enables for each axis.  Won’t be ale to try it until tomorrow.

urshurak776, 2020-09-23:
Downloaded firmware and flashed to board.  Works great.  Nice and quiet!

closed 2020-09-23 by urshurak776