bigtreetech/BIGTREETECH-SKR-mini-E3 #393: Marlin files as downloaded will not compile in VSCode/Platformio

NevilleKB, 2020-09-13:
### Description

I have downloaded the latest version of the SKR Mini E3 files from Github as a ZIP file and extracted the files to my C: drive. Without making any changes I opened the MARLIN-2.0.X-SKR-MINI-E3-V2.0 folder as a PlatformIO project folder and tried to do a Marlin build. It failed

### Steps to reproduce

1. Download the latest files from Github repository
2. Extract to C:\ drive
3. Open MARLIN-2.0.X-SKR-MINI-E3-V2.0  folder in PlatformIO (this folder contains the platformio.ini file)
4. Go to Auto Build Marlin and click on Build
5. Select STM32F103RC_btt_512K as the environment and click build

**Expected behavior**
Should successfully build and produce a firmware.bin file as I have altered nothing

**Actual behavior**
Fails.
<script src="https://pastebin.com/embed_js/XA9EruRM"></script>

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

DexterHill0, 2020-09-13:
I had that error as well. When I was trying to fix it. I found about it [this](https://community.platformio.org/t/marlin-help-me-errror-modulenotfounderror/15586/13). It's something to do with mismatching versions.

To get all my compilation issues fixed, I first reinstalled VSCode and PlatformIO although I'm not sure it's necessary so I would do if below doesn't work. (If you are reinstalling PlatformIO as well as uninstalling it from VSCode, also delete the actual '.platformio' folder.)
For me what worked was getting the latest version of Marlin, and also downloading config files from [here](https://github.com/MarlinFirmware/Marlin/tree/2.0.x/config). To build it I also installed the 'Auto Build Marlin' extension in VSCode.
After doing this I successfully compiled after over a week of no successful compilations.

NevilleKB, 2020-09-14:
Thanks, Dexter, I'll give it a try.

jwm9765, 2020-09-15:
I am having the same issue.
I tried all the steps listed above and I'm still getting the same error.
could you share your files?

DexterHill0, 2020-09-15:
Sure!
Here they are: 
[configs.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5225625/configs.zip)

The changes I made were:
BLTouch
S-Curve Acceleration
Stealthchop
which can be found [https://www.reddit.com/r/ender3/comments/dojh3v/guide_for_those_upgrading_to_an_skr_e3_mini_v12/](here)
I also made a few other changes like disabling the boot screen and other I can't remember. No major major changes were made though.
I am using the latest Marlin version.

DexterHill0, 2020-09-15:
[files.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5226306/files.zip)
What are the errors that you get?

jwm9765, 2020-09-16:
I got all my issues fixed by using your files. I was getting a neopixel error but I fixed it. after 3 days, I finally got it to compile

creatron, 2020-09-19:
Please note: Compile errors is due to bigtreetech not locking version of the platform used, and consequently run afoul of breaking changes in v7.0.0 of the ststm32 platform 13, where almost all boards now default to the stm32duino core, rather than the maple core. This means the other way to fix this would be to change the platform line (#325 above) to read platform = ststm32@6.1.0 which was the latest version at the time they last made changes.  This would solve the issue.
[env:STM32F103RC_btt_512K]
**#platform          = ${common_stm32f1.platform}**
**platform = ststm32@6.1.0** 
extends           = env:STM32F103RC_btt
board_upload.maximum_size=524288
build_flags       = ${env:STM32F103RC_btt.build_flags} -DSTM32_FLASH_SIZE=512 
Please note it is safer to delete the framework -arduinostm32* , platform io will then re-download it and you start fresh..