bigtreetech/BIGTREETECH-SKR-mini-E3 #394: [BUG] Auto Bed Leveling with v2.0.6 failing probing (outside of the bed)

daton89, 2020-09-13:
### Description

I've just installed the Firmware 2.0.6 for Z homing on the e3 mini of my ender 3 pro and the bed leveling fails running outside of the bed. 

I don't find older versions or branches other than master to installer older firmware, have you migrated the repo or what? where can we find older versions of the firmware?

### Steps to reproduce

1. Flash firmware-bltouch-for-z-homing.bin
2. Run Auto Bed Leveling

AnthoPham, 2020-09-13:
I came to also report the same issue. It autohomes perfectly fine and goes off of the bed at the 1st probing point.

chiibis, 2020-09-13:
@daton89 I think you can fix it by setting probe offsets in menu. Did you try it?

AnthoPham, 2020-09-13:
@daton89 I had to do a big of digging but here's the old firmware from 4 months ago that worked for me:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/7300cc6742685716a0bfb59024ad585a626413b4/firmware/V2.0/firmware-bltouch-for-z-homing.bin

daton89, 2020-09-13:
> @daton89 I think you can fix it by setting probe offsets in menu. Did you try it?

@chiibis unfortunately I don't see Probe option for X and Y axes in the menu, only for Z, but it fail completely outside of both X and Y. Other than that it makes crazy noises when the Z axes moves. 

> @daton89 I had to do a big of digging but here's the old firmware from 4 months ago that worked for me:
> https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/7300cc6742685716a0bfb59024ad585a626413b4/firmware/V2.0/firmware-bltouch-for-z-homing.bin

@AnthoPham thank you very much for the suggestion, this older firmware `bugfix 2.0.x` makes it work properly. 

I found also other issues describing similar problems of noise and missing options in the menu so I guess/hope a new firmware version with all this bugfixes is on his way.  ATM I have no time to learn the firmare.

aKarthikprabu, 2020-09-13:
I have the same issue. 

 I have installed SKR MINI E3 V2.0 and 3D Touch V3.0. Updated the firmware from github. However probing is not working. Always it shows probing failed and stays probing 1/9. 

Also when I give Auto Home, the head is not coming to the center. 

When I give G29, the head moves to X stop and stays for few seconds and shows probing failed.

Luckily I purchased SKR mini recently, so had a default firmware which helped me to go back. Now no Auto sensor bed levelling ,,😔

son12710, 2020-09-13:
you can use M851 to change the offset. I had that issue where the the Y offset is set at 10mm

M851 Y-10mm fixed the issue for me. you might need to try different offset number to fit your need

aKarthikprabu, 2020-09-13:
@son12710 , It shows X=10mm and Y=10mm in offset. Z=-2.8 which I set as probe offset. 

While autohome also the head or extruder not comes at center. X=107, Y=107 and Z=12.8

While Bed Levelling, the head stops at X Stop and trying for levelling since it is out of bed, it shows probing failed

son12710, 2020-09-13:
my offset is X-41.47 Y-10.00 Z-1.57 on Ender 3 Pro with BL touch, it worked for me. my board is SKR Mini e3 V2.0
Not sure if it make any different.
when i do auto home, it homes the Z axis in the middle of the bed using the BL Touch ( Z Endstop has been removed)

aKarthikprabu, 2020-09-13:
Can you please explain how did you get negative values in offset because my one shows 10 by default for XY. 

Firmware Location: 4 days before updated as per website
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V2.0

son12710, 2020-09-13:
I uses [https://www.thingiverse.com/thing:3003725](url) for my bltouch. Because the way it was designed. i had to modify the X and Y offset to make the printer prints correctly. 

when i flashed the latest firmware, I noticed that it homed in front of the bed. I was watching so i was able to turn off the printer before it crashed to my bed. At that point, i didn't know what was going on.

then i check my offset and it has X10 Y10. I i then restore my old config which is X-41.47 Y-10.00 (i takes note of everything i do for future references)

aKarthikprabu, 2020-09-14:
Let me try again and come back on this. Thanks for the input.

brelfpv, 2020-09-14:
It's not about the offset.It's not right that the Z axis goes down at the end.It went to the middle and the bed height was measured by the probe, until now.
https://youtu.be/e8JxfKaE7lE

marschmi01, 2020-09-14:
> It's not about the offset.It's not right that the Z axis goes down at the end.It went to the middle and the bed height was measured by the probe, until now.
> https://youtu.be/e8JxfKaE7lE

Remove the jumpers on the board. This fix this Problem.

brelfpv, 2020-09-14:
[image]
Thank you.I just tried pulling out the jumper in two places and it didn't solve the problem.

marschmi01, 2020-09-14:
no, I meant the 4 pieces for the stepper. Not this one. Your Board isn't a V2.0 Board i think!

brelfpv, 2020-09-15:
![Uploading 93153549...]Sorry to complicate the story, but I actually have V2.0 as well, and I still have the same problem with the latest firmware.And this time I switched to the latest firmware for V1.2 and the same error happened.And there was no short clip at the stepper of either board from the beginning.

chiibis, 2020-09-15:
> @chiibis unfortunately I don't see Probe options for X and Y axes in the menu, only for Z, but it fail completely outside of both X and Y. Other than that it makes crazy noises when the Z axes moves.

Its somewhere deep in menu. Just dig a bit and you'll find it. Smth like Configurations > Probe offsets > X-Axis offset

Z-offset has his own menu item on a higher level, but others not.

@daton89

aKarthikprabu, 2020-09-15:
Hi All,
It not a problem in latest firmware. After a long struggle, I found that it is Probe offset issue.
Go to:
Step 1: Configurations
Step 2: Probe offset
Step 3: Set X offset  to -40.00 (for me it worked well)
Step 4: Set Y offset  to -10.00 (for me it worked well)
Step 5: Set Z offset  to -1.70 (for me it worked well)
Step 6: Store the settings.

Thats all and it is done. Everything will work fine. Try the same values as mine except Z offset. It has to be found while leveling.

Use default -2.00 as Z offset then try adjust the values and find the actual value by adjusting the level. Lot of youtube videos are there to know how to set this. If not, send me message and I will reply.

I am also new to 3D printing and will master soon, its not rocket science and need little patience. So enjoy printing.

By the way, I use 3d Touch V3.0 and not BL Touch. I purchased from bangwood and it is quite cheap. It is more than enough for ender 3 pro :)

[image]

brelfpv, 2020-09-16:
For example, in an auto home, after adjusting the XY of the nozzle position with the end sensor, does it come down to the Z offset position at that position?Or do you go to the middle of the bed and then come down?

son12710, 2020-09-16:
@aKarthikprabu yep, just like I said :)

aKarthikprabu, 2020-09-16:
@son12710, thanks.

aKarthikprabu, 2020-09-16:
@brelfpv, is the problem still persisting??? Or did you solve it. Am residing in Dubai, if you are around just ping me, I can support by phone. I know the pain of not working.

brelfpv, 2020-09-17:
> @brelfpv, is the problem still persisting??? Or did you solve it. Am residing in Dubai, if you are around just ping me, I can support by phone. I know the pain of not working.
> 
> 

Thank you.I am Japanese living in Japan.The problem is still unresolved and even the firmware I was tracing from the history is still having problems with the probe's sensor not coming out during auto-home.

aKarthikprabu, 2020-09-17:
Check the sensor wiring and post here if possible, then don't do anything just update latest firmware.

Check bltouch by activating deploy, stow. If working then your sensor and wiring is correct.

brelfpv, 2020-09-17:
Thank you.
I have reverted the firmware back to 2.0.6 and looked for an item to test the probe, but I can't find it anywhere.I took a video of it.
https://youtu.be/vK0RU6j3s3M
[image]

AnthoPham, 2020-09-17:
@daton89 i also had that similar at first (probe powered but could not stow or deploy on autohome)

It came down to wiring. 
See: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/bltouch-2.0-1.jpg

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/bltouch-2.0-2.jpg

To test it, you should be able to go into the configuration menu and deploy/stow the probe. I unfortunately cannot help at the moment as I have 12 hours left on my print.

aKarthikprabu, 2020-09-18:
@brelfpv, based on the picture. Your board is not SKR Mini E3 V2.0 and sensor is not bltouch. It's a clone 3D Touch same as mine. It looks like you are using wrong firmware. Please check according to you board and I guess it is E3 V1.2.

karldonovan, 2020-09-18:
This version is not great, thought I would give it a go and used the z homing (to check an issue with the parts cooling fan). This version seems slow and really noisy in comparison to the older version. I also fail probing as it is off the bed. Who is writing this rubbish at BTT? Went back to the older version and out of curiosity came to see if anyone else was experiencing an issue.

AnthoPham, 2020-09-18:
> 
> 
> @brelfpv, based on the picture. Your board is not SKR Mini E3 V2.0 and sensor is not bltouch. It's a clone 3D Touch same as mine. It looks like you are using wrong firmware. Please check according to your board and I guess it is E3 V1.2.
> 
> [image]

> 
> 
> Sorry to complicate the story, but I actually have V2.0 as well, and I still have the same problem with the latest firmware.And this time I switched to the latest firmware for V1.2 and the same error happened.And there was no short clip at the stepper of either board from the beginning.[image]

@aKarthikprabu, he previously attached a photo of his V2.0 version stating that he had the same issue on both boards.

@brelfpv, I linked previously the BL touch connection and firmware that worked for me for the V2.0 board. 

> 
> 
> I had to do a big of digging but here's the old firmware from 4 months ago that worked for me:
> https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/7300cc6742685716a0bfb59024ad585a626413b4/firmware/V2.0/firmware-bltouch-for-z-homing.bin

I'm not familiar with the V1.2 board and so I'm unsure of the connections being correct there but my fix with the firmware from above was the connections which I've added in another reply. 

> 
> 
> i also had that similar at first (probe powered but could not stow or deploy on autohome)
> 
> It came down to wiring as I had something similar to what you've shown before with the same issue
> 
> See: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/bltouch-2.0-1.jpg
> 
> https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/bltouch-2.0-2.jpg
> 
> To test it, you should be able to go into the configuration menu and deploy/stow the probe. I unfortunately cannot help at the moment as I have 12 hours left on my print.

With the V2.0 board, the BL Touch (or probe settings) commands were found under Configuration > Bl Touch. From there you should be able to diagnose if your wiring is correct. 

Best of luck

brelfpv, 2020-09-19:
Thanks, I'll check the wiring.

brelfpv, 2020-09-19:
Sorry, I asked about the 1.2v issue in relation to asking about the 2.0v issue and the conversation expanded.I have both.And now I'm talking about 1.2.

brelfpv, 2020-09-19:
I think the wiring is the same connection that has worked in the past, and I have checked it a few times since this issue occurred, but I will check it again.
So I simply want to ask, when the auto home with zhoming firmware, does it move to the setting Z on the spot after the nozzle touches the X and Y sensors at the end of the bed?
or will the nozzle never come down before the nozzle have gone to the middle of the bed, if only? I think it strange that it comes down before it goes to the middle.

TheDarthGrader, 2020-09-21:
> Hi All,
> It not a problem in latest firmware. After a long struggle, I found that it is Probe offset issue.
> Go to:
> Step 1: Configurations
> Step 2: Probe offset
> Step 3: Set X offset to -40.00 (for me [...]

THANK YOU! This has been driving me crazy. Being on a Mac didn't help.

Silversunbeam, 2020-09-21:
This is most definitely a firmware issue. My machine was running fine with the previous firmware release. X/Y Offset in all previous releases was fine for those using the Creality supplied BLTouch mount. In addition, the stepper drivers seem to be off kilter somehow as the X and Y axis motors were making an unusual whine. I ended up reverting to the previous release to regain operability.

brelfpv, 2020-09-22:
I agree with that.

IvSan, 2020-09-23:
If you think that it's a firmware issue related to Marlin itself - take a look at this https://github.com/MarlinFirmware/Marlin/issues/19487

ramiabraham, 2020-10-06:
> Hi All,
> It not a problem in latest firmware. After a long struggle, I found that it is Probe offset issue.
> Go to:
> Step 1: Configurations
> Step 2: Probe offset
> Step 3: Set X offset to -40.00 (for me [...]

I'm having the same issue as OP, and can't get this solution to work - does anyone know how to access the settings under _Configuration -> Advanced Settings -> Set Home Offsets_? I always receive an error when attempting to set offsets. NO matter what I've tried so far, it goes like this:

- Initially attempt to access the settings under _Configuration -> Advanced Settings -> Set Home Offsets_
- Printer returns to info screen and prints the error _"Home XYZ First" _
- Attempt to home X, Y, and Z via the _Motion -> Auto Home_ option (I've also tried homing each axis individually)
- Printer returns to info screen and prints the error _"Err: Too Far!" _

- Ender 3
- SKR mini E3 V2 board
- genuine bltouch
- Creality bltouch mount
- For the firmware, I've tried both _firmware-bltouch.bin_ and _firmware-bltouch-for-z-homing.bin_

Currently on the printer is _firmware-bltouch-for-z-homing.bin_ @ Marlin 2.0.6.1.

Thanks, any suggestions appreciated

alexnrobv, 2020-10-09:
Hello, i am currently facing the same situation as @ramiabraham, i have the ANTCLABS BLTOUCH 3.1 and SKR mini V2 with TFT 35 on an Ender 3 Pro
I have managed to set the offsets for the bed in the Probe offset menu, but it seems that for the Z it does not work as expected. 

I have connected both the Z-end Stop and BL Touch to the board as per the pinout described, everything is working correctly, i can test, i can use all the commands. 
When i try to Auto Bed Level it homes first using the end stops on all axes then goes to the center, lifts Z 10 mm. and goes to the   X and Y coordinates, i have set in the Probe Offset menu. And then starts probing, resulting in error.

If i try to change the offset to Z-10 for example then the lift after homing will be 20 mm. but if i set it to 9.8 for example, it will still lift to 10 mm.

I currently have the firmware-bltouch.bin on the printer.

Thank you kindly.