bigtreetech/BIGTREETECH-SKR-mini-E3 #395: Bootloader's for all current production boards

LearnIIBurn, 2020-09-13:
Hello,

In light of a bunch of the issues popping up regarding not being able to flash off SD and blue screens and "bricked" boards, is it possible for us to have access to the bootloader standalone for each board? I own quite a few BTT SKR boards and all of them seem to be in some state of broken or kind of working or just plain 🗑️🔥

I'm about to get an ST-Link and focus on getting my printer working like it was months ago which right now feels pretty far to complete.

Thanks to anyone at BTT or in the community that has bootloader's to share.

abacabie, 2020-09-15:
i think That's a good idea, there certainly seems to be few people with these issues and this may well help.

LearnIIBurn, 2020-09-22:
Well regardless of if they want to share bootloaders or not I just got that ST-Link and BTT sent me instructions in Chinese on how to flash the bootloader on Windows so I'm going to use google translate on the doc and try to do it on Mac. Should be no big deal.

fraserw, 2020-09-22:
FYI I was having these problems as well. Turns out the issues were entirely with how github allows one to download the precompiled firmwares. Some of the ways one can download from links on the git pages cause files to download with the wrong size. They should be about 230 kb, but some come in at just over 100 kb.

From the version directory, click on the filename of the firmware of interest. That will direct to a nearly blank page, containing "view raw". Click the download button to the upper right. 

That was the only way I was able to get the full corret file without cloning the entire repo locally.

gioeleminardi, 2020-09-23:
> Well regardless of if they want to share bootloaders or not I just got that ST-Link and BTT sent me instructions in Chinese on how to flash the bootloader on Windows so I'm going to use google translate on the doc and try to do it on Mac. Should be no big deal.

Would you like to share the instructions from BTT with us?

LearnIIBurn, 2020-09-29:
@gioeleminardi Sure thing!

[stlink引导程序烧录方法.docx](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5299282/stlink.docx)

Thats what they sent me. Basically hook up your St-Link to SWD pins in correct order to BTTs pinout and have a Windows machine to work on (I struggled for a long time trying to get any ST software to start on Mac OS Catalina) and use St-Link Utility to connect to the board and read/write and all that. They gave me a link to smoothieware bootloader as the bootloader for the E3-Mini-v2.0. https://github.com/Smoothieware/Smoothieware/tree/edge/bootloader

LearnIIBurn, 2020-09-29:
> FYI I was having these problems as well. Turns out the issues were entirely with how github allows one to download the precompiled firmwares. Some of the ways one can download from links on the git [...]

Wow I think you may be on to something here. I noticed some pretty small .bin files I've downloaded http from github. Very interesting indeed.

Switchcrafter, 2020-10-19:
downloading specific files and the difference in size or function is a github thing.  Right-clicking and Save as.. or pressing the download button from the specific file will cause the issue.   

The best way to grab files is either 
(A.) cloning the repository locally
(B.)  hitting that green CODE button in the main repository then downloading the ZIP

lcortese, 2023-04-13:
hey guy! does anyone know where is the binary bootloader for "srk mini e3 v1.2"?
thanks!