bigtreetech/BIGTREETECH-SKR-mini-E3 #396: SKR Mini E3 V2.0

Tandy70, 2020-09-14:
Bonjour, SKR Mini E3 V2.0 sur une ender3 , la carte SD contenant le firmware n'est pas flasher , tout les fichiers firmware.bin essayer restent en .bin pas .cur. l'imprimante n'est pas reliée . j'ai essayer avec la carte de l'écran TFT 35 et L'USB non plus . Ne se connecte pas avec l'ordinateur et Pronterface .
Un grand MERCI pour une solution . Salutations

Marco-Santos20, 2020-09-30:
Good afternoon my name is Marco Santos I have an SKR MINI E3 V2.0 bord I have all the cables well fitted in the right quote but the fan where the SKR MINI E3 V 2.0 bord is not working I would like what I can do for it to work

Tandy70, 2020-09-30:
Bonsoir Monsieur Santos , comment avez vous trouver mon adresse? J’ai aussi une carte SKR E3 Mini V2.0 , le lecteur SD ne fonctionne pas, j’attends une réponse du vendeur . J’ai aussi une carte mère SKR 1.3 et une 1.4 Turbo qui ne fonctionne pas. Du même vendeur officiel Bigtreetech . Je peut pas vous aider , j’ai vue sur internet beaucoup de clients avec ce problème, si vous avez un écran LCD avec Marlin affiche souvent la cause du problème. Vérifier votre firmware . Salutations 

Envoyé par myMail pour iOS