bigtreetech/BIGTREETECH-SKR-mini-E3 #398: Mini 3.0 V1.2 Marlin will not compile.

Jschoepp, 2020-09-19:
### Description
I installed the board into an Ender 5. The supplied firmware loaded but the directions for the stepper motors and the positions for the end stops are different in the Ender 5. I need to reverse them. 
Marlin compilation will stop with a many errors when using your Marlin firmware for SKR Mini 3.0 V 1.2. See screenshot.

### Steps to reproduce

1. Loading Marlin environment from the folder into VS Code does not work. 
2. Used download from the Official Marlin website and copy the config files from BIGTREETECH over, update platformio with STM32F103RC_btt_512K
3. Compile starts but crashes.
4. Tried to use generic Marlin latest rev and modify the platformio and config files. It compiles correctly but when I put it into the SD of the board it will not read in. -> blank screen. 

**Expected behavior**
Good Compilation. 

**Actual behavior**
Compilation crashes. 

### Additional Information
[image]

Arakon, 2020-09-20:
The errors in red tell you exactly what the issue is. You have to change the functon "EVENT_GCODE_SD_STOP" to "EVENT_GCODE_SD_ABORT" in either configuration.h or configuration_adv.h.

Avramih, 2020-09-21:
If you use Official Marlin, so take config files from it, not from BIGTREETECH. There is examples, look for Creality/Ender-3/BigTreeTech SKR Mini E3 1.2.
https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20Mini%20E3%201.2