bigtreetech/BIGTREETECH-SKR-mini-E3 #399: [BUG] Motor directions reversed

fraserw, 2020-09-21:
### Description

All  motors are running in reverse direction. 

### Steps to reproduce

1. Direct swap of creality default board to the SKR mini V2
2. Install ender 5 firmware from BTT repo. Direct bin install, no local compilation. Install using SD card
3. Run an auto home

**Expected behavior**

One by one all axes should move towards their limit switches

**Actual behavior**

One by one, all axes move the wrong way before slamming into the frame.

### Additional Information

Ender 5 pro. Using the firmware bin files provided by BTT directly.

jwg4096, 2020-09-22:
I concur, same issue here.

RedCoder09, 2020-09-22:
So, an easy way to fix this is to get firmware and follow a guide, and then you want to change these from false to true:

#define INVERT_X_DIR
#define INVERT_Y_DIR
#define INVERT_Z_DIR

Hope this helps!

fraserw, 2020-09-22:
You mean, to compile the firmware. Yeah I agree. But based on the other bug reports, compiling the current rev of the firmware is borked. So I haven't bothered getting into it yet.

Another easy fix is in the hardware - reverse the pins of the drive stepper motors. Did that last night and all seems well. Will have to swap back of course, once a new firmware rev is produced. Easy enough.

RedCoder09, 2020-09-22:
Ah, so you probably should close the issue.

fraserw, 2020-09-22:
But the issue -- a bug in the firmware -- is not solved. All I have provided is a workaround solution that is not a permanent solution, as it requires hardware modifications to switch between different versions of marlin, or firmware compiled by others.

This issue is not solved

fraserw, 2020-09-22:
Some additional details - tried my first print through octoprint, and the motor directions are reversed, even after my hardwire pin reversal. So it seems the reversed motor directions are only with control from the LCD screen, and through octoprint, they are correct.

I don't get it.

RedCoder09, 2020-09-23:
Hmm... It's possible the inverts in the firmware were automatically set to true for some reason.

fraserw, 2020-09-24:
The updates to the compiled firmware seem to have made the motor reversal consistent at least. 

Updated to the recent using firmware_ender5.bin. Autohoming from LCD produces reversed motor movement, as does control and printing through octopi. 

Reversed motor pins on the x, y, and z axes allow for an operable patch in the interim.

Link3M, 2020-09-24:
Hi, i'm new here, i bought skr mini 3 v2 and updated the firmware but home is going to the left instead of right, checked cable positions and everything is in right position, 
Is there a way to fix it?
I'm not much into programming or coding, is there a tutorial i can follow so i can resolve it by myself or if this is going to be correct or if it has been corrected already is there a link where to download the latest firmware? 

Thank You
=)

fraserw, 2020-09-24:
There is no tutorial that I know of. But as I stated in my previous message, if you simply reverse the pins in the stepper motor wires at either the mobo or drive ends, for the X, Y, and Z motors, the firmware  submitted on Sept 23 will work perfectly fine. It'll take you less than an hour.

Once they update the firmware with a proper fix for the direction reversal, you'll have to reverse the pins in the connectors again.

thebiggestgit, 2020-12-05:
I don't get this issue but slightly different re X and Y Axis. I have just updated my Ender 5 Pro (with BL Touch) with the SKR Mini v2 board with the SKR firmware-ender5-bltouch-for-z-homing.bin firmware Marlin 2.0.7.2.  The BL Touch auto-homing (after swaping blue and red cables over), bed levelling and printing work fine (through Octoprint - not tested from printer itself), although the print is rotated or possibly mirrored on the bed (will check which with a new print).

thebiggestgit, 2020-12-05:
On my ender 5 Pro the y and x axis were similarly reversed, I set on the touch screen to reverse the x and y axis which worked, although the prints are now rotated 180 degrees around on the bed, so the x/y 0,0 is now opposite the original when using the stock creality 8 bit board.  So this isn't actually a problem as it prints and levels, though not quite right in their firmware which I think may be for the ender 3 bed still? This is using the latest firmware with BL touch firmware-ender5-bltouch-for-z-homing.bin Marlin 2.0.7.2.  Only just installed the board so no idea about previous version behaviour.

thebiggestgit, 2020-12-07:
I had this same issue with my Ender 5 Pro (with BL touch and using the Z-Stop with that) with the x and y axis being reversed after replacing the stock 8 bit Creality board.  As a workaround I changed this via the touch screen to invert x and y which allowed it to print with the latest SKR Ender 5 BL Touch Z-home firmware (firmware-bltouch-for-z-homing-ender5.bin), however the bed was rotated 180 degrees, so the origin was in opposite corner, but prints were OK otherwise (but not via Octoprint).  Ended up recompiling their SKR Marlin firmware myself to fix this and a few other tweaks like fixing USB Serial connection timeouts (increasing buffer) so I could print from Octoprint reliably, which the precompiled firmware above didn't do.  So far, seems OK...

cengbrecht, 2022-01-10:
The fact that this is still an issue for a product this mature (In terms of years) is crazy. 
That being said, this issue still exists, and is not 100% consistent. 
I reversed the XY and Z motors, and now the Z is reversed, I just re-reversed it to see if that resolves, but even with them reversed, the motors still go to the wrong corner?

fraserw, 2022-01-27:
Do you mean go to the wrong corner during homing, during start or end of prints, or something else?

cengbrecht, 2022-01-27:
Home was set to 220x220x0
I have fixed that, but the motors are still run backwards.

fraserw, 2022-01-27:
Yep I agree. Don't consider it a deal breaker as it was an easy fix.

Hey would you mind sharing your config files? I haven't had a need to compile my own yet, but I will quite soon. Would be nice to have a good place to start!

cengbrecht, 2022-01-27:
Hahaha, I have other issues, I wouldn't recommend using mine. :P 
However, I have a .h, and adv.h that you could have, though I have NOT tried them yet.

fraserw, 2022-01-27:
I'd appreciate it. Mind posting here?

cengbrecht, 2022-01-27:
[Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/7953346/Marlin.zip)

BoogerX, 2022-04-10:
Just upgraded my Ender 5 Plus and seeing the same issue X/Y/Z all going in opposite directions. Is there a firmware fix for this or are we still swapping motor pins?

Brosk1, 2022-06-09:
For Info, just installed the SKR E3 V3 in my Ender 5 Pro, and the XYZ were inverted compared to the 1.1.5 that was in there, that was with the stock Ender3 that was pre loaded in the board. I flashed with 2.0.9.3 which corrected the Z but if I tried a Auto Home the X was headed for doomsville, I rebooted and it corrected itself, all Axis including the DD upgrade is the right direction. had to calc Esteps, but that's it. Now comes the fun of calibration etc etc. - not my code but this is where its from - https://codeload.github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/zip/refs/heads/master