bigtreetech/BIGTREETECH-SKR-mini-E3 #400: SKR Mini E3 V2.0 wont boot

SirRonBird, 2020-09-21:
Hello,

my skr mini e3 v2.0 wont boot... I´ve installed the board and wired up everything correctly (triple-checked it). On first startup everything went fine, so I wrote a new Firmware following this guide: https://www.reddit.com/r/ender3/comments/h8y1ia/marlin_20x_guide_skr_mini_e3_v20_ender_3/

I put the firmeware.bin onthe SD card and put it in the Printer and turned it on... nothing happend just the blue light on the Screen. I waited about 5 minutes before I turned the Power off.

Now everytime I turn the power on the screen still stays blue and nothing happend. The power-led is on and the fans are moving...

Is the Bootloader broken or did i something wrong? I also tried to flash this firmware: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V2.0/firmware-bltouch.bin it dindt help.

PLS HELP ME

SirRonBird, 2020-09-22:
> [#224 (comment)](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/224#issuecomment-695179614)

Can I use this on the Stock LCD-Display ?

getz97, 2020-09-22:
no,
in the Sd card which come with skr mini e3 v2.0, have a original""FIRMWARE.bin""
use it.

SirRonBird, 2020-09-25:
I figuered it out.... the Board was defect I ordered a new one and it work fine

szzs84, 2020-10-10:
I've had tha same issue. Probably the board is not faulty. When I downloaded the firmware.bin directly from github, it was only about 110kB. When You pull the whole project az a ZIP, extract is, the firmware.bin is about 215kB. Flash it, and everything will be fine.