bigtreetech/BIGTREETECH-SKR-mini-E3 #405: TMC CONNECTION ERROR

rockguy902, 2020-09-23:
I am trying to connect my printer via USB to windows 10 and use pronterface and I can get the printer connected but NOTHING ELSE HAPPENS. I was once able to move the print head left and right on the Y axis but thats all it wont home or anything.

I have since tried to use a different firmware of my own compiling and I cant get VSCode to actually create a file for me to put on my SD card and the printer wont connect to the PC so VSCode doesnt recognize the printer my LCD screen only ever says TMC CONNECTION ERROR and then eventually the connection is killed and I have to shut down the printer. what if anything am I doing wrong

jamesbretz, 2020-09-24:
Make sure you are actually powering the printer from the power supply, USB power only will cause this issue.

TheTelz, 2020-09-24:
I too am seeing this error. After firmware upgrade I get TMC CONNECTION ERROR. I am powering printer via power supply that came with Ender 3 Pro.

rockguy902, 2020-09-24:
> I too am seeing this error. After firmware upgrade I get TMC CONNECTION ERROR. I am powering printer via power supply that came with Ender 3 Pro.

So i learned to power on the printer first let it fully boot,  then plug it to the pc

TheTelz, 2020-09-24:
> > I too am seeing this error. After firmware upgrade I get TMC CONNECTION ERROR. I am powering printer via power supply that came with Ender 3 Pro.
> 
> So i learned to power on the printer first let it fully boot, then plug it to the pc

Yeah I always do this. So the precompiled firmware seems to work but if I compile my own, leaving everything default I get the TMC issue. I'm betting the precompiled has something configured that the config files in the branch don't.

TheTelz, 2020-09-24:
I was able to resolve the issue. Within the Configuration.h file there are lines #define X_DRIVER_TYPE  TMC2209. I changed it to 2208 and now TMC issue is gone. 

#define X_DRIVER_TYPE  TMC2208
#define Y_DRIVER_TYPE  TMC2208
#define Z_DRIVER_TYPE  TMC2208

#define E0_DRIVER_TYPE TMC2208

ntaft, 2020-09-25:
This appears to be broken with the latest commit, which made a number of configuration changes (including reducing the build volume to 230 from 235mm and disabling bed leveling). 

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/commit/4997b6165d6e5cf6fed39bb93e1ebcaf6e5ff7cb#diff-1101bd8b21f60bcbd0311a056503d49e

The Y, Z and E0 slave addresses were all changed from their default at 0, which seemed to disconnect them. Reverting this change fixed the issue for me.

Trevor-Miller, 2020-09-27:
Confirmed, reverting the slave addresses to 0 was the only change that needed to be made to fix the issue

aerialviews007, 2020-09-28:
I can confirm.  Reverting those addresses to 0 fixed the issue for me.  What is the impact of this change?

felda, 2021-01-21:
I am currently having this issue happen when turning the printer off, but still leaving USB plugged in. I tried setting the slave addresses all to zero, but Auto Compile Marlin plugin for VS Code would not compile afterwards.

Is the only fix to have the USB unplugged or powered off when powering off the printer and making sure the printer powers on first before the USB device?