bigtreetech/BIGTREETECH-SKR-mini-E3 #406: m999 error bl touch ender 3 pro

jimthedinosuar, 2020-09-24:
Hi, I recently bought a bl touch 3.1 and installed it it came up with the m999 error and wasn't sure why. I tried rerouting the wires but it didn't work. Thank you if you have a solution.

oldman4U, 2020-09-25:
Please close the ticket in the SKR Pro repository. 

Thank you

empakoso, 2020-09-30:
My problem was the bltouch wiring.  Check it against this:

https://www.antclabs.com/wiring
And then accordingly to this ( I prefer this method) 

[image]

empakoso, 2020-09-30:
This is also my reference https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/BTT%20SKR%20MINI%20E3%20V2.0-PIN.pdf

jimthedinosuar, 2020-10-01:
Thank you @empakoso I will give it a try